stm32duino/Arduino_Core_STM32 #1309: STM32F1 UART TX idles after every byte

jlpoltrack, 2021-02-25:
**Describe the bug**
When using UART1 on STM32F1, TX seem to idle after every byte sent.  Don't expect this idle period. 

**To Reproduce**
Source from Arduino IDE:

`void setup() {
  Serial.begin(1000000);
}

void loop() {
  Serial1.printf("Test");
  Serial1.flush();
  delay(100);
}`

Screenshot from Scope:
[image]

**Expected behavior**
Expect to see continuous output from TX, such as:

[
[image]
](url)

**Screenshots**
Screenshots above

**Desktop (please complete the following information):**
 - OS: Win 10
 - Arduino IDE version: 1.8.13
 - STM32 core version: 1.9.0
 - Tools menu settings if not the default: Default
 - Upload method: Serial

**Board (please complete the following information):**
 - Name: STM32F103C Blue Pill

ABOSTM, 2021-03-03:
Hi @jlpoltrack,
As I understand, there is no real issue, you just don't get the full performances.
I don't consider this behavior has a bug, it works the way it has been implemented:
Current implementation send each bytes one by one, even if a multi character string is provided as argument.
And the consequence is that there is interrupt and CPU activity between each bytes (to prepare and send the next bytes).

Nevertheless, I agree that it is not really optimized.
This is why I propose a slightly different implementation in PR #1313

jlpoltrack, 2021-03-05:
@ABOSTM 
Thanks for the response and explaining the current implementation.  As I didn't see this mentioned in the API documentation, thought I had something wrong with my setup.  I noticed when I saw a regression in packet size at higher UART speeds.  
I look forward to trying out the new implementation in the future.

PR #1313 (2021-03-10): HardwareSerial: don't call STM32 HAL for every bytes to transfer

**Summary**

HardwareSerial: don't call STM32 HAL for every bytes to transfer

Instead of calling HAL for each and every bytes,
try to call it only once for a string.
Several call may be needed to manage internal circular buffer (tx_buff)

Fixes #1309
closed 2021-03-10 by fpistm