stm32duino/Arduino_Core_STM32 #1315: Arduino_Core_STM32 version 1.9.0 regression?

rei-vilo, 2021-03-06:
I'm using **Arduino_Core_STM32** version 1.9.0 on Arduino IDE 1.83, and trying to blink the LED of the Nucleo STM32F401RE with the standard Blink example. The firmware version is V2J37M27. 

The sketch compiles, links, uploads but the green LED doesn't blink.

I've tried different upload options, mass storage, Serial, SWD, to no avail.

I downgraded **Arduino_Core_STM32** to version 1.8.0 and everything runs as expected. The green LED blinks. 

Is there a regression on Arduino_Core_STM32 version 1.9.0?

fpistm, 2021-03-06:
Which board did you select in the menu?

rei-vilo, 2021-03-06:
I've selected 

- Board: `Nucleo-64`, and then 
- Board part number: `"Nucleo F401RE"`

[image]

fpistm, 2021-03-07:
The main difference btw 1.8.0 and 1.9.0 for this board is the systemClock_config which has been update to use the HSE bypass instead of the HSI.

rei-vilo, 2021-03-07:
Thanks for the explanation. So it seems we have identified the cause of the regression.

Have you managed to replicate the issue with the blinky example?

fpistm, 2021-03-08:
>  So it seems we have identified the cause of the regression.

No, I just point the main change. 

> Have you managed to replicate the issue with the blinky example?

Yes and it works fine on my side.
If you have an issue I guess you have an issue on your side but difficult to tell which one.
Is your board modified?
You plugged only the USB cable on the STLink?

Maybe your the HSE bypass provided by the STLink is not correct. You can try to override the clock config by adding in your sketch:

```C
extern "C" void SystemClock_Config(void)
{
  RCC_OscInitTypeDef RCC_OscInitStruct = {};
  RCC_ClkInitTypeDef RCC_ClkInitStruct = {};

  /* Configure the main internal regulator output voltage */
  __HAL_RCC_PWR_CLK_ENABLE();
  __HAL_PWR_VOLTAGESCALING_CONFIG(PWR_REGULATOR_VOLTAGE_SCALE2);
  /* Initializes the CPU, AHB and APB busses clocks */
  RCC_OscInitStruct.OscillatorType = RCC_OSCILLATORTYPE_HSE;
  RCC_OscInitStruct.HSEState = RCC_HSE_BYPASS;
  RCC_OscInitStruct.PLL.PLLState = RCC_PLL_ON;
  RCC_OscInitStruct.PLL.PLLSource = RCC_PLLSOURCE_HSE;
  RCC_OscInitStruct.PLL.PLLM = 8;
  RCC_OscInitStruct.PLL.PLLN = 336;
  RCC_OscInitStruct.PLL.PLLP = RCC_PLLP_DIV4;
  RCC_OscInitStruct.PLL.PLLQ = 7;
  if (HAL_RCC_OscConfig(&RCC_OscInitStruct) != HAL_OK) {
    Error_Handler();
  }
  /* Initializes the CPU, AHB and APB busses clocks */
  RCC_ClkInitStruct.ClockType = RCC_CLOCKTYPE_HCLK | RCC_CLOCKTYPE_SYSCLK
                                | RCC_CLOCKTYPE_PCLK1 | RCC_CLOCKTYPE_PCLK2;
  RCC_ClkInitStruct.SYSCLKSource = RCC_SYSCLKSOURCE_PLLCLK;
  RCC_ClkInitStruct.AHBCLKDivider = RCC_SYSCLK_DIV1;
  RCC_ClkInitStruct.APB1CLKDivider = RCC_HCLK_DIV2;
  RCC_ClkInitStruct.APB2CLKDivider = RCC_HCLK_DIV1;

  if (HAL_RCC_ClockConfig(&RCC_ClkInitStruct, FLASH_LATENCY_2) != HAL_OK) {
    Error_Handler();
  }
}

```

rei-vilo, 2021-03-08:
The Nucleo-F401RE is unmodified with version MB1136 C-01.

I've already used the `SystemClock_Config()` previously, and checked it again, to no avail.

So I am keeping release 1.8.0 to target the Nucleo-F401RE.

Thank you for your help.

fpistm, 2021-03-08:
OK. So I've found why it does not work on your side:

```
The board marking MB1136 C-01 corresponds to a board, configured as HSE not used.
The board marking MB1136 C-02 (or higher) corresponds to a board, configured to use
ST-LINK MCO as the clock input.
```
The C 01 is not configured to used the HSE. Using the `SystemClock_Config` above will alow you to use the 1.9.0.

closed 2021-03-08 by fpistm

rei-vilo, 2021-03-09:
Thank you, it works as expected now.