stm32duino/Arduino_Core_STM32 #1316: STM32L432 eeprom working?

JyeSmith, 2021-03-07:
Hey,

I am trying to use eeprom on the STM32L432.  It should be pretty straight forward but I dont appear to be having any luck.  Is someone able to confirm it works?

Cheers

fpistm, 2021-03-07:
Hi @JyeSmith 
could you be more precise ?
Which board? Arduino IDE version? Core version ?
Which sketch?

Basically the EEPROM library is an emulation using last page flash.

fpistm, 2021-03-09:
I've tested `eeprom_put `and `eeprom_get `default examples and it works fine on the Nucleo L432KC

cruwaller, 2021-03-18:
@fpistm this is not working as expected on L432 when flash size is smaller than max 256kB :)

Current code calculates flash page index:
`#define FLASH_PAGE_NUMBER   ((uint32_t)((FLASH_SIZE / FLASH_PAGE_SIZE) - 1))` 
=> FLASH_SIZE returns 0x20000 and page is 0x800 => page 63 (the address `0x0801F800`). 

Then EEPROM base address is defined:
`#define FLASH_BASE_ADDRESS  ((uint32_t)((FLASH_END + 1) - FLASH_PAGE_SIZE))`
=> `#define FLASH_END             (0x0803FFFFUL)` is defined in `stm32l432xx.h` => `0x0803FFFFUL` => base address will be `0x803F800` which is actually page 127 :)

This means that EEPROM save will work just once after mass erase because wrong flash page is erased :)

Could you reopen this error?

fpistm, 2021-03-18:
@ABOSTM could you have a look on this please? Thanks in advance.

ABOSTM, 2021-03-18:
@JyeSmith,
I agree with you, it is a real issue.
FLASH_END which is defined in CMSIS file doesn't take into account the real flash size of a the device,
(it is just the maximum possible within the  productline).

But fix will be a real challenge, as other STM32 series are impacted, and it will be difficult to find generic implementation, taking into account single bank/dual bank, possible flash discontinuity between banks, ...
So as a workaround, I propose you to update, for now, stm32l432xx.h with the value you expected:
`#define FLASH_END (0x0801FFFFUL)`

Kamoool, 2021-05-12:
I cannot save to EEPROM as well on STM32L432KBU6 on latest version.
On 1.8.0 both EEPROM.put and EEPROM.get worked.

PR #1445 (2021-07-13): fix: library: EEPROM: use correct flash size
`FLASH_SIZE` defined by the CMSIS Device Peripheral Access Layer Header File is the maximum one of a product line.
So depending of the mcu this is not the correct one.

As an example, L432KC have 256 KB while L432Kb has 128 KB.

Using `LL_GetFlashSize()` allows to get the correct FLASH size.

Fixes #1316
commit f4f0f817 (2021-07-13): fix: library: EEPROM: use correct flash size

FLASH_SIZE defined by the CMSIS Device Peripheral Access Layer Header File
is the maximum one of a product line.
So depending of the mcu this is not the correct one.

As an example, L432KC have 256 KB while L432Kb has 128 KB.

Using LL_GetFlashSize() allows to get the correct FLASH size.

Fixes #1316
closed 2021-07-13 by fpistm