stm32duino/Arduino_Core_STM32 #1317: Version 1.9.0 -  Header file core_cm3.h is missing

kevintedder, 2021-03-07:
I've just install version 1.9.0 on the Arduino IDE (v1.8.13) and run a compile.  It throws out this error message:
 C:\Users\kevin\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.9.0\system/Drivers/CMSIS/Device/ST/STM32F1xx/Include/stm32f103x6.h:127:10: fatal error: core_cm3.h: No such file or directory
  127 | #include "core_cm3.h"
      |          ^~~~~~~~~~~~

File Explorer search finds nothing.    I assume this repo would download and install all the necessary files to compile successfully.     What am I doing wrong?

Can someone please advise me as to where/how to download this?

Many  thx.
kevin

fpistm, 2021-03-07:
Hi @kevintedder 
the `core_cm3.h` is provided by the CMSIS package installed with the core. If you don't have it you probably had an issue during the installation.
Try to reinstall.

kevintedder, 2021-03-07:
Thanks. The second install worked.

closed 2021-03-07 by kevintedder