stm32duino/Arduino_Core_STM32 #1323: STM32F405RG Not Working

harry1453, 2021-03-11:
**Describe the bug**
Code does not run at all on Generic STM32F405RG target.

**To Reproduce**
```c++
#define LED_PIN PA13
void setup() {
    pinMode(LED_PIN , OUTPUT);
}

void loop() {
    digitalWrite(LED_PIN , LOW);
    delay(500);
    digitalWrite(LED_PIN , HIGH);
   delay(500);
  }
}
```

Steps to reproduce the behavior:
1. Upload
2. Nothing happens.

**Expected behavior**
The LED blinks.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Windows]
 - Arduino IDE version: 1.8.12
 - STM32 core version: 1.9.0
 - Tools menu settings if not the default:
[image]
 - Upload method: DFU

**Board (please complete the following information):**
 - Name: Generic STM32F405RG (It's a [Matek Flight Controller](http://www.mateksys.com/?portfolio=f405-wing))

**Additional context**
When I switch to Adafruit Feather F405, or when I include the following updated timer code from `master`, the issue gets weirder.
```c++
/*
 * @brief  Configures the System clock source, PLL Multiplier and Divider factors,
 *               AHB/APBx prescalers and Flash settings
 * @note   This function should be called only once the RCC clock configuration
 *         is reset to the default reset state (done in SystemInit() function).
 * @param  None
 * @retval None
 */

/******************************************************************************/
/*            PLL (clocked by HSE) used as System clock source                */
/******************************************************************************/
static uint8_t SetSysClock_PLL_HSE(uint8_t bypass)
{
  RCC_OscInitTypeDef RCC_OscInitStruct;
  RCC_ClkInitTypeDef RCC_ClkInitStruct;

  /* The voltage scaling allows optimizing the power consumption when the device is
  clocked below the maximum system frequency, to update the voltage scaling value
  regarding system frequency refer to product datasheet. */
  __HAL_RCC_PWR_CLK_ENABLE();
  __HAL_PWR_VOLTAGESCALING_CONFIG(PWR_REGULATOR_VOLTAGE_SCALE2);

  // Enable HSE oscillator and activate PLL with HSE as source
  RCC_OscInitStruct.OscillatorType      = RCC_OSCILLATORTYPE_HSE;
  if (bypass == 0) {
    RCC_OscInitStruct.HSEState          = RCC_HSE_ON; // External 8 MHz xtal on OSC_IN/OSC_OUT
  } else {
    RCC_OscInitStruct.HSEState          = RCC_HSE_BYPASS; // External 8 MHz clock on OSC_IN
  }

  RCC_OscInitStruct.PLL.PLLState        = RCC_PLL_ON;
  RCC_OscInitStruct.PLL.PLLSource       = RCC_PLLSOURCE_HSE;
  RCC_OscInitStruct.PLL.PLLM            = HSE_VALUE / 1000000L; // Expects an 8 MHz external clock by default. Redefine HSE_VALUE if not
  RCC_OscInitStruct.PLL.PLLN            = 336;                  // VCO output clock = 336 MHz (1 MHz * 336)
  RCC_OscInitStruct.PLL.PLLP            = RCC_PLLP_DIV2;        // PLLCLK = 168 MHz (336 MHz / 2)
  RCC_OscInitStruct.PLL.PLLQ            = 7;
  if (HAL_RCC_OscConfig(&RCC_OscInitStruct) != HAL_OK) {
    return 0; // FAIL
  }

  // Select PLL as system clock source and configure the HCLK, PCLK1 and PCLK2 clocks dividers
  RCC_ClkInitStruct.ClockType = RCC_CLOCKTYPE_SYSCLK | RCC_CLOCKTYPE_HCLK | RCC_CLOCKTYPE_PCLK1 | RCC_CLOCKTYPE_PCLK2;
  RCC_ClkInitStruct.SYSCLKSource   = RCC_SYSCLKSOURCE_PLLCLK;
  RCC_ClkInitStruct.AHBCLKDivider  = RCC_SYSCLK_DIV1;         // 168 MHz
  RCC_ClkInitStruct.APB1CLKDivider = RCC_HCLK_DIV4;           // 42 MHz
  RCC_ClkInitStruct.APB2CLKDivider = RCC_HCLK_DIV2;           // 84 MHz
  if (HAL_RCC_ClockConfig(&RCC_ClkInitStruct, FLASH_LATENCY_5) != HAL_OK) {
    return 0; // FAIL
  }

  /* Output clock on MCO1 pin(PA8) for debugging purpose */
  /*
  if (bypass == 0)
    HAL_RCC_MCOConfig(RCC_MCO1, RCC_MCO1SOURCE_HSE, RCC_MCODIV_2); // 4 MHz
  else
    HAL_RCC_MCOConfig(RCC_MCO1, RCC_MCO1SOURCE_HSE, RCC_MCODIV_1); // 8 MHz
  */

  return 1; // OK
}

/******************************************************************************/
/*            PLL (clocked by HSI) used as System clock source                */
/******************************************************************************/
uint8_t SetSysClock_PLL_HSI(void)
{
  RCC_OscInitTypeDef RCC_OscInitStruct;
  RCC_ClkInitTypeDef RCC_ClkInitStruct;

  /* The voltage scaling allows optimizing the power consumption when the device is
    clocked below the maximum system frequency, to update the voltage scaling value
    regarding system frequency refer to product datasheet. */
  __HAL_RCC_PWR_CLK_ENABLE();
  __HAL_PWR_VOLTAGESCALING_CONFIG(PWR_REGULATOR_VOLTAGE_SCALE2);

  // Enable HSI oscillator and activate PLL with HSI as source
  RCC_OscInitStruct.OscillatorType      = RCC_OSCILLATORTYPE_HSI | RCC_OSCILLATORTYPE_HSE;
  RCC_OscInitStruct.HSIState            = RCC_HSI_ON;
  RCC_OscInitStruct.HSEState            = RCC_HSE_OFF;
  RCC_OscInitStruct.HSICalibrationValue = RCC_HSICALIBRATION_DEFAULT;
  RCC_OscInitStruct.PLL.PLLState        = RCC_PLL_ON;
  RCC_OscInitStruct.PLL.PLLSource       = RCC_PLLSOURCE_HSI;
  RCC_OscInitStruct.PLL.PLLM            = 16;            // VCO input clock = 1 MHz (16 MHz / 16)
  RCC_OscInitStruct.PLL.PLLN            = 336;           // VCO output clock = 336 MHz (1 MHz * 336)
  RCC_OscInitStruct.PLL.PLLP            = RCC_PLLP_DIV2; // PLLCLK = 168 MHz (336 MHz / 2)
  RCC_OscInitStruct.PLL.PLLQ            = 7;
  if (HAL_RCC_OscConfig(&RCC_OscInitStruct) != HAL_OK) {
    return 0; // FAIL
  }

  /* Select PLL as system clock source and configure the HCLK, PCLK1 and PCLK2 clocks dividers */
  RCC_ClkInitStruct.ClockType      = (RCC_CLOCKTYPE_SYSCLK | RCC_CLOCKTYPE_HCLK | RCC_CLOCKTYPE_PCLK1 | RCC_CLOCKTYPE_PCLK2);
  RCC_ClkInitStruct.SYSCLKSource   = RCC_SYSCLKSOURCE_PLLCLK;
  RCC_ClkInitStruct.AHBCLKDivider  = RCC_SYSCLK_DIV1;         // 168 MHz
  RCC_ClkInitStruct.APB1CLKDivider = RCC_HCLK_DIV4;           // 42 MHz
  RCC_ClkInitStruct.APB2CLKDivider = RCC_HCLK_DIV2;           // 84 MHz
  if (HAL_RCC_ClockConfig(&RCC_ClkInitStruct, FLASH_LATENCY_5) != HAL_OK) {
    return 0; // FAIL
  }

  /* Output clock on MCO1 pin(PA8) for debugging purpose */
  //HAL_RCC_MCOConfig(RCC_MCO1, RCC_MCO1SOURCE_HSI, RCC_MCODIV_1); // 16 MHz

  return 1; // OK
}

void SystemClock_Config(void)
{
  /* 1- If fail try to start with HSE and external xtal */
  if (SetSysClock_PLL_HSE(0) == 0) {
    /* 2- Try to start with HSE and external clock */
    if (SetSysClock_PLL_HSE(1) == 0) {
      /* 3- If fail start with HSI clock */
      if (SetSysClock_PLL_HSI() == 0) {
        Error_Handler();
      }
    }
  }

  /* Ensure CCM RAM clock is enabled */
  __HAL_RCC_CCMDATARAMEN_CLK_ENABLE();

  /* Output clock on MCO2 pin(PC9) for debugging purpose */
  //HAL_RCC_MCOConfig(RCC_MCO2, RCC_MCO2SOURCE_SYSCLK, RCC_MCODIV_4);
}
```
Now, **the LED blinks once and then doesn't blink ever again**. This is really confusing me - it seems like the MCU crashes or something. I can't easily debug as the USB serial doesn't work (Device Descriptor Request Failed). I have tried both wrapping it in another `for` loop and using `while (millis() - start > 500);` as an alternative to `delay()` but it always blinks once, no matter what the delays, and then crashes.

harry1453, 2021-03-12:
I have just updated my core to the latest master. Here's a table of what works and what doesn't:

Target | With clock code | Without clock code
-- | -- | --
Feather | Works! | LED is extremely dim, but works.
Generic | Blinks once | Blinks once

harry1453, 2021-03-12:
I've been playing with the core trying to get it to work, and I've got to the point where even after reverting the core to 1.9.0, I can't get it to work anymore, even with the strange workarounds I found earlier. I am truly stuck, unfortunately this bug makes this core useless on this chip :(

fpistm, 2021-03-12:
Hi @harry1453 
Generic variant try first to use an HSE  then HSE bypass and finally HSI.
My guess is that the HSE or HSE by pass failed.Is your board has an external oscillator?

Note: can be linked to this issue: https://github.com/STMicroelectronics/STM32CubeF4/issues/17

ABOSTM, 2021-03-12:
Hi @harry1453,
What I propose you is to  check the different steps below:

**1) Work with HSI as System clock (No PLL) 16MHz :**
in your sketch add an empty SystemClock_Config. This will keep default clock configuration after reset (aka HSI) 
```
void SystemClock_Config(void)
{
} 
```
Check if led is blinking correctly. If ok, then check step2
  
  
**2) Work with PLL as SystemClock (168MHz) with HSI as source**
In your sketch: 
```
void SystemClock_Config(void)
{
  RCC_OscInitTypeDef RCC_OscInitStruct = {0};
  RCC_ClkInitTypeDef RCC_ClkInitStruct = {0};

  /** Configure the main internal regulator output voltage
  */
  __HAL_RCC_PWR_CLK_ENABLE();
  __HAL_PWR_VOLTAGESCALING_CONFIG(PWR_REGULATOR_VOLTAGE_SCALE1);
  /** Initializes the RCC Oscillators according to the specified parameters
  * in the RCC_OscInitTypeDef structure.
  */
  RCC_OscInitStruct.OscillatorType = RCC_OSCILLATORTYPE_HSI;
  RCC_OscInitStruct.HSIState = RCC_HSI_ON;
  RCC_OscInitStruct.HSICalibrationValue = RCC_HSICALIBRATION_DEFAULT;
  RCC_OscInitStruct.PLL.PLLState = RCC_PLL_ON;
  RCC_OscInitStruct.PLL.PLLSource = RCC_PLLSOURCE_HSI;
  RCC_OscInitStruct.PLL.PLLM = 8;
  RCC_OscInitStruct.PLL.PLLN = 168;
  RCC_OscInitStruct.PLL.PLLP = RCC_PLLP_DIV2;
  RCC_OscInitStruct.PLL.PLLQ = 4;
  if (HAL_RCC_OscConfig(&RCC_OscInitStruct) != HAL_OK)
  {
    Error_Handler();
  }
  /** Initializes the CPU, AHB and APB buses clocks
  */
  RCC_ClkInitStruct.ClockType = RCC_CLOCKTYPE_HCLK|RCC_CLOCKTYPE_SYSCLK
                              |RCC_CLOCKTYPE_PCLK1|RCC_CLOCKTYPE_PCLK2;
  RCC_ClkInitStruct.SYSCLKSource = RCC_SYSCLKSOURCE_PLLCLK;
  RCC_ClkInitStruct.AHBCLKDivider = RCC_SYSCLK_DIV1;
  RCC_ClkInitStruct.APB1CLKDivider = RCC_HCLK_DIV4;
  RCC_ClkInitStruct.APB2CLKDivider = RCC_HCLK_DIV2;

  if (HAL_RCC_ClockConfig(&RCC_ClkInitStruct, FLASH_LATENCY_5) != HAL_OK)
  {
    Error_Handler();
  }
}
```

Check if led is blinking correctly.

ABOSTM, 2021-03-12:
@harry1453 Don't forget the `extern "C"` like I did initially (thanks @fpistm)

harry1453, 2021-03-12:
Thanks for the response. I've tried both options. Interestingly removing/adding `extern "C"` makes no difference in either case, which is suspicious to me.

First case (empty clock config) = LED lights very dimly, blinks off, and then doesn't come back on.

Second case = LED blinks, albeit too quickly and very dimly.

@fpistm How can I tell which STM32Cube version I'm using to make sure that I'm on 1.26.0 to avoid that bug?

I'm trying to find out the clock configuration of my board but unfortunately it is a closed source design and I'm struggling to find it... It seems most likely to me that it has an external oscillator, but I'm not sure. All other similar flight controller boards I can find information on all have external 8MHz oscillators, so this seems most likely.

ABOSTM, 2021-03-12:
Even if I don't understand, I confirm that with or without `extern "C"` produce the same result: the  `SystemClock_Config()` function used is the one of the sketch.
Now I am very surprised that with empty clock configuration it doesn't work. Could you try the same (step1 and step2) with another led ? I mean on another gpio pin

ABOSTM, 2021-03-12:
Looking at http://www.mateksys.com/?portfolio=f405-wing#tab-id-3 I have the feeling that LED is not a simple led, but rather a Led strip (which imply completely different control).

fpistm, 2021-03-12:
> @fpistm How can I tell which STM32Cube version I'm using to make sure that I'm on 1.26.0 to avoid that bug?

https://github.com/stm32duino/Arduino_Core_STM32/blob/master/system/Drivers/STM32YYxx_HAL_Driver_version.md

harry1453, 2021-03-12:
Regarding the board's LED pad, that's for Adafruit NeoPixel LEDs. I'm using the built-in one marked LED1 here (circled):
[image]
Unfortunately I don't have the schematic for this board so I can't tell if it's just an LED. I only [know](https://github.com/iNavFlight/inav/blob/master/src/main/target/MATEKF405SE/target.h#L24) that it's on `PA13`.

However...

I changed my program to this:
```
#include <Arduino.h>

void setup() {
    pinMode(PA10, OUTPUT);
}

void loop() {
    digitalWrite(PA10, LOW);
    delay(500);
    digitalWrite(PA10, HIGH);
    delay(500);
}

extern "C" void SystemClock_Config(void)
{
}
```

and on `PA10` I get the following:
[image]
....which is perfect!!! I have no idea why the LED doesn't work as expected... especially when it worked with the Feather F405 variant.. but I'll have to debug that later.

With the 168MHz PLL clock the code also works correctly.

However, when removing the function entirely the default implementation is still broken - no output is seen on the pin.

harry1453, 2021-03-13:
Sorry, but I'm now struggling to get USB VCP to work... Could the issue be to do with the timer configuration?

Using PIO,
```
build_flags =
    -D USBD_USE_CDC
    -D USBCON
    -D HAL_PCD_MODULE_ENABLED
```
and code:
```
void setup() {
    SerialUSB.begin();
    delay(10000);
    SerialUSB.println("Hello World!");
}

void loop() {
}

extern "C" void SystemClock_Config(void)
{
  (Code from above)
}
```
The only result is Device Descriptor Request Failed. I know that this board is good for USB VCP because other firmwares use it on this board. Declaring `-D USE_USB_HS` causes it to not even be detected as a USB device.

ABOSTM, 2021-03-15:
Hi @harry1453,
The "SystemClock_Config(void)" above doesn't configure USB clock properly, thus USB could not work.

Below is the modified code (PLLQ modified).
Please note that USB with HSI based clock is not optimal, it would be better to use HSE (if you succeed to get information of the HSE on your board)
```
extern "C" void SystemClock_Config(void)
{
  RCC_OscInitTypeDef RCC_OscInitStruct = {0};
  RCC_ClkInitTypeDef RCC_ClkInitStruct = {0};

  /** Configure the main internal regulator output voltage
  */
  __HAL_RCC_PWR_CLK_ENABLE();
  __HAL_PWR_VOLTAGESCALING_CONFIG(PWR_REGULATOR_VOLTAGE_SCALE1);
  /** Initializes the RCC Oscillators according to the specified parameters
  * in the RCC_OscInitTypeDef structure.
  */
  RCC_OscInitStruct.OscillatorType = RCC_OSCILLATORTYPE_HSI;
  RCC_OscInitStruct.HSIState = RCC_HSI_ON;
  RCC_OscInitStruct.HSICalibrationValue = RCC_HSICALIBRATION_DEFAULT;
  RCC_OscInitStruct.PLL.PLLState = RCC_PLL_ON;
  RCC_OscInitStruct.PLL.PLLSource = RCC_PLLSOURCE_HSI;
  RCC_OscInitStruct.PLL.PLLM = 8;
  RCC_OscInitStruct.PLL.PLLN = 168;
  RCC_OscInitStruct.PLL.PLLP = RCC_PLLP_DIV2;
  RCC_OscInitStruct.PLL.PLLQ = 7;
  if (HAL_RCC_OscConfig(&RCC_OscInitStruct) != HAL_OK)
  {
    Error_Handler();
  }
  /** Initializes the CPU, AHB and APB buses clocks
  */
  RCC_ClkInitStruct.ClockType = RCC_CLOCKTYPE_HCLK|RCC_CLOCKTYPE_SYSCLK
                              |RCC_CLOCKTYPE_PCLK1|RCC_CLOCKTYPE_PCLK2;
  RCC_ClkInitStruct.SYSCLKSource = RCC_SYSCLKSOURCE_PLLCLK;
  RCC_ClkInitStruct.AHBCLKDivider = RCC_SYSCLK_DIV1;
  RCC_ClkInitStruct.APB1CLKDivider = RCC_HCLK_DIV4;
  RCC_ClkInitStruct.APB2CLKDivider = RCC_HCLK_DIV2;

  if (HAL_RCC_ClockConfig(&RCC_ClkInitStruct, FLASH_LATENCY_5) != HAL_OK)
  {
    Error_Handler();
  }
}

```

ABOSTM, 2021-03-15:
By the way, we tested Adafruit Feather F405, and led is blinking properly on PA13.

harry1453, 2021-03-15:
Thank you for your time helping with this. I don't know what's going on with PA13's LED. PA14's LED works completely fine and since I don't have a board schematic I'll have to ignore that issue for now.

I've managed to figure out that my board has a 12MHz HSE. So, the following code works for an LED blink:

```
extern "C" void SystemClock_Config(void)
{
    RCC_OscInitTypeDef RCC_OscInitStruct = {0};
    RCC_ClkInitTypeDef RCC_ClkInitStruct = {0};

    /** Configure the main internal regulator output voltage
    */
    __HAL_RCC_PWR_CLK_ENABLE();
    __HAL_PWR_VOLTAGESCALING_CONFIG(PWR_REGULATOR_VOLTAGE_SCALE1);
    /** Initializes the RCC Oscillators according to the specified parameters
    * in the RCC_OscInitTypeDef structure.
    */
    RCC_OscInitStruct.OscillatorType = RCC_OSCILLATORTYPE_HSE;
    RCC_OscInitStruct.HSEState = RCC_HSE_ON;
    RCC_OscInitStruct.PLL.PLLState = RCC_PLL_ON;
    RCC_OscInitStruct.PLL.PLLSource = RCC_PLLSOURCE_HSE;
    RCC_OscInitStruct.PLL.PLLM = 12;
    RCC_OscInitStruct.PLL.PLLN = 168;
    RCC_OscInitStruct.PLL.PLLP = RCC_PLLP_DIV2;
    RCC_OscInitStruct.PLL.PLLQ = 4;
    if (HAL_RCC_OscConfig(&RCC_OscInitStruct) != HAL_OK)
    {
        Error_Handler();
    }
    /** Initializes the CPU, AHB and APB buses clocks
    */
    RCC_ClkInitStruct.ClockType = RCC_CLOCKTYPE_HCLK|RCC_CLOCKTYPE_SYSCLK
                                  |RCC_CLOCKTYPE_PCLK1|RCC_CLOCKTYPE_PCLK2;
    RCC_ClkInitStruct.SYSCLKSource = RCC_SYSCLKSOURCE_PLLCLK;
    RCC_ClkInitStruct.AHBCLKDivider = RCC_SYSCLK_DIV1;
    RCC_ClkInitStruct.APB1CLKDivider = RCC_HCLK_DIV4;
    RCC_ClkInitStruct.APB2CLKDivider = RCC_HCLK_DIV2;

    if (HAL_RCC_ClockConfig(&RCC_ClkInitStruct, FLASH_LATENCY_5) != HAL_OK)
    {
        Error_Handler();
    }
}
```

Unfortunately, both the clock config you sent and the above code with `RCC_OscInitStruct.PLL.PLLQ = 4;` still fails to initialize USB :(

harry1453, 2021-03-15:
Thank you for your time helping with this. I don't know what's going on with PA13's LED. PA14's LED works completely fine and since I don't have a board schematic I'll have to ignore that issue for now.

I probed the board and found that the HSE is 8MHz. After studying the PLL configuration some more (this is my first time programming STM32...) I came up with the following:

```
RCC_OscInitStruct.OscillatorType = RCC_OSCILLATORTYPE_HSE;
RCC_OscInitStruct.HSEState = RCC_HSE_ON;
RCC_OscInitStruct.PLL.PLLState = RCC_PLL_ON;
RCC_OscInitStruct.PLL.PLLSource = RCC_PLLSOURCE_HSE;
RCC_OscInitStruct.PLL.PLLM = 4; // VCO Input Clock 2MHz
RCC_OscInitStruct.PLL.PLLN = 168; // VCO Output Clock 336MHz
RCC_OscInitStruct.PLL.PLLP = RCC_PLLP_DIV2; // Core Clock 168MHz
RCC_OscInitStruct.PLL.PLLQ = 7; // USB Clock 48MHz....
```

This works! :)

fpistm, 2021-03-15:
So if all is fine now. This issue can be closed?

harry1453, 2021-03-15:
Yes, thank you both very much for your help!

closed 2021-03-15 by harry1453

JyeSmith, 2021-07-01:
Hi @harry1453 

I came across this issue due to the same problem and a simple update fixed it.

But Im having difficulties with serial output and thought Id ask you first before opening a new issue because Im also using a FC.

Below is the .ini and main.  The LED blinks fine, but no matter what I try to do with serial and setting pins I cant get an output.

Serial upload works so I know the uart is ok.  Have you gotten print to work?

Cheers

```
[env:genericSTM32F405RG]
platform = ststm32@14.0.1
framework = arduino
board = genericSTM32F405RG
upload_protocol = serial
```

```
#include <Arduino.h>

#define LEDPIN PA8

void setup()
{
  Serial.setTx(PA9);
  Serial.setRx(PA10);
  Serial.begin(9600);

  pinMode(LEDPIN, OUTPUT);
}

void loop()
{
  Serial.println("loop");

  digitalWrite(LEDPIN, HIGH);
  delay(250);
  digitalWrite(LEDPIN, LOW);
  delay(250);
}
```

uzi18, 2021-07-02:
@JyeSmith try this: https://github.com/stm32duino/wiki/wiki/API#hardwareserial

JyeSmith, 2021-07-03:
@uzi18 I gave it another go and no output.

Serial upload worked fine and then no output.

```
[env:genericSTM32F405RG]
platform = ststm32@14.0.1
framework = arduino
board = genericSTM32F405RG
upload_protocol = serial
```