stm32duino/Arduino_Core_STM32 #1325: Could not change PWM resolution in pins_arduino.h

michael172363, 2021-03-14:
Hi,
I am having an issue with setting the PWM resolution.

Tried to change on my STM3F407VGT6 board the PWM settings to 10kHz frequency and 12 bits resolution.

So I have edited the settings in pins_arduino.h file.

The PWM frequency was running on 10kHz afterwards, but the resolution was still the same.
Tried to initialize additionally analogWriteResolution(12); in the setup function, but still no change.

On the oscilloscope I measured that the pulse increases in steps of every 0.4µs. From a total pulse length of 100µs.

**Desktop**
 - OS: Windows 10
 - VS Code + PlatformIO
 - Framework: arduinoststm32 4.10900.200819 (1.9.0)
 - Platform: ST STM32 (10.0.1) > Black STM32F407VG
 - Upload method: STLink

**Board**
 - Name: DevEBox STM32F407VGT6

**main.cpp**
```
#include <Arduino.h>
int pwmvalue = 0;

void setup()
{
    analogWriteResolution(12);
}

void loop()
{

    analogWrite(PA0, pwmvalue);
    delay(100);
    pwmvalue += 1;
    if (pwmvalue >= 4095)
    {
        pwmvalue = 0;
    }
}
```

**pins_arduino.h**
File location:
C:\Users\Michael\.platformio\packages\framework-arduinoststm32\cores\arduino

```
#ifndef ADC_RESOLUTION
  #define ADC_RESOLUTION              12 
#endif

#define DACC_RESOLUTION             12

#ifndef PWM_RESOLUTION
 #define PWM_RESOLUTION              12

```

**platformio.ini**
```
[env:black_f407vg]
platform = ststm32
board = black_f407vg
framework = arduino
upload_protocol = stlink
```

[image]
[image]

ABOSTM, 2021-03-15:
Hi @michael172363,
There is no need to modify `pins_arduino.h`, you just have to call both API below that in your sketch (in `setup()`)
```
  analogWriteResolution(12);
  analogWriteFrequency(10000);
```

I tested with BluePill (STM32F103C8T6),
(with both above API call)
 I slightly modified your sketch to force a specific value of `pwmvalue`:
* when pwmvalue = 1 then pulse length around 20ns  (here I am a bit at the limit of my analyser, so value is a bit rounded)
* when pwmvalue = 2 then  pulse length around 40ns
*  ....
* when pwmvalue = 10 then  pulse length around 235ns (here I overpass analyser limit, and I can compare to theoretical value)

This correspond to 12Bit resolution: (100µs /4096) = 24,4ns
The same with my nucleo f429zi.

Can you try the same: set pwmvalue = 10 and measure the pulse ?

michael172363, 2021-03-15:
Hi @ABOSTM,
with pwmvalue = 10, I got around 238 ns.
Timing of the other values I tried is also quite similar to yours.

Seems like it is working! Thanks a lot for your help!

closed 2021-03-15 by michael172363