stm32duino/Arduino_Core_STM32 #1327: Issues with new Arduino IDE 2.0 please get involved

toogooda, 2021-03-15:
**Describe the bug**
The new Arduino IDE is in Beta testing 3 and I have been raising problems I am finding. They may need changes to boards.txt etc to ensure this core continues to work with the official Arduino IDE.

Ports all messed up just for your core
https://github.com/arduino/arduino-ide/issues/225
[image]

Items showing as unidentified
https://github.com/arduino/arduino-ide/issues/220
[image]

**To Reproduce**
Complete source code which can be used to reproduce the issue. Please try to be as generic as possible (no extra code, extra hardware,...)

Steps to reproduce the behavior:
See links to issues

**Desktop (please complete the following information):**
 - Windows10
 - Arduino IDE version: [2.0 Beta3]
 - STM32 core version: 1.9

fpistm, 2021-03-16:
Hi @toogooda 
Thanks for reporting this anyway it is linked to Arduino IDE 2.0.0 and honestly I don't know how to help.

toogooda, 2021-03-16:
> Hi @toogooda
> Thanks for reporting this anyway it is linked to Arduino IDE 2.0.0 and honestly I don't know how to help.

The CLI returns the following when you plug in any F103 variant (any I have):

```
c:\Program Files (x86)\Arduino_CLI>arduino-cli board list --format json
[
  {
    "address": "COM5",
    "protocol": "serial",
    "protocol_label": "Serial Port (USB)",
    "boards": [
      {
        "name": "Maple Mini",
        "FQBN": "stm32duino:STM32F1:mapleMini",
        "VID": "0x1EAF",
        "PID": "0x0004"
      },
      {
        "name": "Maple (RET6)",
        "FQBN": "stm32duino:STM32F1:mapleRET6",
        "VID": "0x1EAF",
        "PID": "0x0004"
      },
      {
        "name": "Generic STM32F103C series",
        "FQBN": "stm32duino:STM32F1:genericSTM32F103C",
        "VID": "0x1EAF",
        "PID": "0x0004"
      },
      {
        "name": "Generic STM32F103T series",
        "FQBN": "stm32duino:STM32F1:genericSTM32F103T",
        "VID": "0x1EAF",
        "PID": "0x0004"
      },
      {
        "name": "Generic STM32F103Z series",
        "FQBN": "stm32duino:STM32F1:genericSTM32F103Z",
        "VID": "0x1EAF",
        "PID": "0x0004"
      },
      {
        "name": "Microduino Core STM32 to Flash",
        "FQBN": "stm32duino:STM32F1:microduino32_flash",
        "VID": "0x1EAF",
        "PID": "0x0004"
      },
      {
        "name": "Generic STM32F103R series",
        "FQBN": "stm32duino:STM32F1:genericSTM32F103R",
        "VID": "0x1EAF",
        "PID": "0x0004"
      },
      {
        "name": "HYTiny STM32F103TB",
        "FQBN": "stm32duino:STM32F1:hytiny-stm32f103t",
        "VID": "0x1EAF",
        "PID": "0x0004"
      },
      {
        "name": "Maple (Rev 3)",
        "FQBN": "stm32duino:STM32F1:maple",
        "VID": "0x1EAF",
        "PID": "0x0004"
      },
      {
        "name": "Generic STM32F103V series",
        "FQBN": "stm32duino:STM32F1:genericSTM32F103V",
        "VID": "0x1EAF",
        "PID": "0x0004"
      }
    ]
  }
]
```
So the unique VID/PID is duplicated on many devices, Doesn't the mapping for VID/PID to FQBN come from your core? They seem to all use the same VID/PID I assumed the fix was to have  "VID": "0x1EAF" and "PID": "0x0004" map to something generic like "STM32F103 Series"... 
Just keen to make sure STM remains a big part of Arduino. But don't really know too much about the inner workings of the cores.

fpistm, 2021-03-16:
The list you provide comes from this core: https://github.com/rogerclarkmelbourne/Arduino_STM32
Not this one 🤣

toogooda, 2021-03-16:
> The list you provide comes from this core: https://github.com/rogerclarkmelbourne/Arduino_STM32
> Not this one 🤣

Interesting as I never installed that one on IDE 2.0, I used it years ago on the old IDE. So old cores installed on the old IDE still effect the new CLI...
Can you please provide how you verified the above so I can update the bug.

Thanks

fpistm, 2021-03-17:
>  Can you please provide how you verified the above so I can update the bug.

Simply check the boards.txt you will see that information comes from the one from https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/9ef6d9def84c998b7e590c6b5f27a5a796632e44/STM32F1/boards.txt#L13-L14
Moreover the FQBN indicates it is arch STM32F1: `"FQBN": "stm32duino:STM32F1:mapleMini"` while this core it is "stm32".
So I close this issue as it is not link to this core.

closed 2021-03-17 by fpistm

toogooda, 2021-03-18:
Thanks yes I found an old install of that core, once removed that issue  went away. thanks for your help :)