stm32duino/Arduino_Core_STM32 #1328: Unable to use "stm32l0xx_ll_tim.h" directly or via "stm32yyxx_ll_tim.h"

LukeGary462, 2021-03-15:
Hello,

I am prototyping some software using the stm32duino core and the NUCLEO-L073RZ board. 
I am able to compile and upload code from the HAL API but not the simpler LL API. I have used both in IAR, but I cannot seem to 
configure the build system to properly link the stm32l0xx_ll_tim.o file.  From the output in the arduino output dialog, it seems that each of the files I am including are actually being compiled, they are just not getting linked into the final output.

Here is a snippet of my init code:
```c++
#include "stm32yyxx_ll_tim.h"
#include "stm32l0xx_ll_tim.h"
#include "stm32l0xx_hal.h"
#include "stm32l0xx_hal_gpio.h"
#include <Arduino.h>

void setup(){
    LL_TIM_InitTypeDef tim_config;
    LL_TIM_ENCODER_InitTypeDef encoder_config;

    
    (void)LL_TIM_StructInit(&tim_config);
    tim_config.Prescaler = 3u;
    tim_config.Autoreload = 10000u;
    (void)LL_TIM_Init(TIM22, &tim_config);

    (void)LL_TIM_ENCODER_StructInit(&encoder_config);
    (void)LL_TIM_ENCODER_Init(TIM22, &encoder_config);
}
```

And the output from the arduino software:
```bash
C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\hardware -hardware D:\Documents\ArduinoData\packages -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\tools-builder -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\hardware\tools\avr -tools D:\Documents\ArduinoData\packages -built-in-libraries C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\libraries -libraries D:\Documents\Arduino\libraries -fqbn=STM32:stm32:Nucleo_64:pnum=NUCLEO_L073RZ,upload_method=MassStorage,xserial=generic,usb=none,xusb=FS,opt=osstd,rtlib=nano -vid-pid=0483_374B -ide-version=10813 -build-path C:\Users\luke\AppData\Local\Temp\arduino_build_385493 -warnings=none -build-cache C:\Users\luke\AppData\Local\Temp\arduino_cache_409817 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xpack-arm-none-eabi-gcc.path=D:\Documents\ArduinoData\packages\STM32\tools\xpack-arm-none-eabi-gcc\9.2.1-1.1 -prefs=runtime.tools.xpack-arm-none-eabi-gcc-9.2.1-1.1.path=D:\Documents\ArduinoData\packages\STM32\tools\xpack-arm-none-eabi-gcc\9.2.1-1.1 -prefs=runtime.tools.CMSIS.path=D:\Documents\ArduinoData\packages\STM32\tools\CMSIS\5.5.1 -prefs=runtime.tools.CMSIS-5.5.1.path=D:\Documents\ArduinoData\packages\STM32\tools\CMSIS\5.5.1 -prefs=runtime.tools.STM32Tools.path=D:\Documents\ArduinoData\packages\STM32\tools\STM32Tools\1.4.0 -prefs=runtime.tools.STM32Tools-1.4.0.path=D:\Documents\ArduinoData\packages\STM32\tools\STM32Tools\1.4.0 -verbose D:\Documents\Arduino\stm32l073-control-prototype\stm32l073-control-prototype.ino
C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\arduino-builder -compile -logger=machine -hardware C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\hardware -hardware D:\Documents\ArduinoData\packages -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\tools-builder -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\hardware\tools\avr -tools D:\Documents\ArduinoData\packages -built-in-libraries C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\libraries -libraries D:\Documents\Arduino\libraries -fqbn=STM32:stm32:Nucleo_64:pnum=NUCLEO_L073RZ,upload_method=MassStorage,xserial=generic,usb=none,xusb=FS,opt=osstd,rtlib=nano -vid-pid=0483_374B -ide-version=10813 -build-path C:\Users\luke\AppData\Local\Temp\arduino_build_385493 -warnings=none -build-cache C:\Users\luke\AppData\Local\Temp\arduino_cache_409817 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xpack-arm-none-eabi-gcc.path=D:\Documents\ArduinoData\packages\STM32\tools\xpack-arm-none-eabi-gcc\9.2.1-1.1 -prefs=runtime.tools.xpack-arm-none-eabi-gcc-9.2.1-1.1.path=D:\Documents\ArduinoData\packages\STM32\tools\xpack-arm-none-eabi-gcc\9.2.1-1.1 -prefs=runtime.tools.CMSIS.path=D:\Documents\ArduinoData\packages\STM32\tools\CMSIS\5.5.1 -prefs=runtime.tools.CMSIS-5.5.1.path=D:\Documents\ArduinoData\packages\STM32\tools\CMSIS\5.5.1 -prefs=runtime.tools.STM32Tools.path=D:\Documents\ArduinoData\packages\STM32\tools\STM32Tools\1.4.0 -prefs=runtime.tools.STM32Tools-1.4.0.path=D:\Documents\ArduinoData\packages\STM32\tools\STM32Tools\1.4.0 -verbose D:\Documents\Arduino\stm32l073-control-prototype\stm32l073-control-prototype.ino
Using board 'Nucleo_64' from platform in folder: D:\Documents\ArduinoData\packages\STM32\hardware\stm32\1.9.0
Using core 'arduino' from platform in folder: D:\Documents\ArduinoData\packages\STM32\hardware\stm32\1.9.0
"D:\\Documents\\ArduinoData\\packages\\STM32\\tools\\STM32Tools\\1.4.0/tools/win/busybox.exe" sh "D:\\Documents\\ArduinoData\\packages\\STM32\\hardware\\stm32\\1.9.0\\system/extras/prebuild.sh" "C:\\Users\\luke\\AppData\\Local\\Temp\\arduino_build_385493" "D:\\Documents\\Arduino\\stm32l073-control-prototype"
Detecting libraries used...
"D:\\Documents\\ArduinoData\\packages\\STM32\\tools\\xpack-arm-none-eabi-gcc\\9.2.1-1.1/bin/arm-none-eabi-g++" -mcpu=cortex-m0plus -mthumb "@C:\\Users\\luke\\AppData\\Local\\Temp\\arduino_build_385493/sketch/build_opt.h" -c -Os -w -std=gnu++14 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -fno-use-cxa-atexit "-ID:\\Documents\\Arduino\\stm32l073-control-prototype" "-ID:\\Documents\\ArduinoData\\packages\\STM32\\hardware\\stm32\\1.9.0\\cores\\arduino/avr" "-ID:\\Documents\\ArduinoData\\packages\\STM32\\hardware\\stm32\\1.9.0\\cores\\arduino/stm32" "-ID:\\Documents\\ArduinoData\\packages\\STM32\\hardware\\stm32\\1.9.0\\cores\\arduino/stm32/LL" "-ID:\\Documents\\ArduinoData\\packages\\STM32\\hardware\\stm32\\1.9.0\\cores\\arduino/stm32/usb" "-ID:\\Documents\\ArduinoData\\packages\\STM32\\hardware\\stm32\\1.9.0\\cores\\arduino/stm32/OpenAMP" "-ID:\\Documents\\ArduinoData\\packages\\STM32\\hardware\\stm32\\1.9.0\\cores\\arduino/stm32/usb/hid" "-ID:\\Documents\\ArduinoData\\packages\\STM32\\hardware\\stm32\\1.9.0\\cores\\arduino/stm32/usb/cdc" "-ID:\\Documents\\ArduinoData\\packages\\STM32\\hardware\\stm32\\1.9.0\\system/Drivers/STM32L0xx_HAL_Driver/Inc" "-ID:\\Documents\\ArduinoData\\packages\\STM32\\hardware\\stm32\\1.9.0\\system/Drivers/STM32L0xx_HAL_Driver/Src" "-ID:\\Documents\\ArduinoData\\packages\\STM32\\hardware\\stm32\\1.9.0\\system/STM32L0xx" "-ID:\\Documents\\ArduinoData\\packages\\STM32\\hardware\\stm32\\1.9.0\\system/Middlewares/ST/STM32_USB_Device_Library/Core/Inc" "-ID:\\Documents\\ArduinoData\\packages\\STM32\\hardware\\stm32\\1.9.0\\system/Middlewares/ST/STM32_USB_Device_Library/Core/Src" "-ID:\\Documents\\ArduinoData\\packages\\STM32\\hardware\\stm32\\1.9.0\\system/Middlewares/OpenAMP" "-ID:\\Documents\\ArduinoData\\packages\\STM32\\hardware\\stm32\\1.9.0\\system/Middlewares/OpenAMP/open-amp/lib/include" "-ID:\\Documents\\ArduinoData\\packages\\STM32\\hardware\\stm32\\1.9.0\\system/Middlewares/OpenAMP/libmetal/lib/include" "-ID:\\Documents\\ArduinoData\\packages\\STM32\\hardware\\stm32\\1.9.0\\system/Middlewares/OpenAMP/virtual_driver" -w -x c++ -E -CC -DSTM32L0xx -DARDUINO=10813 -DARDUINO_NUCLEO_L073RZ -DARDUINO_ARCH_STM32 "-DBOARD_NAME=\"NUCLEO_L073RZ\"" -DSTM32L073xx -DHAL_UART_MODULE_ENABLED -D__CORTEX_SC=0 "-ID:\\Documents\\ArduinoData\\packages\\STM32\\tools\\CMSIS\\5.5.1/CMSIS/Core/Include/" "-ID:\\Documents\\ArduinoData\\packages\\STM32\\hardware\\stm32\\1.9.0\\system/Drivers/CMSIS/Device/ST/STM32L0xx/Include/" "-ID:\\Documents\\ArduinoData\\packages\\STM32\\hardware\\stm32\\1.9.0\\system/Drivers/CMSIS/Device/ST/STM32L0xx/Source/Templates/gcc/" "-ID:\\Documents\\ArduinoData\\packages\\STM32\\tools\\CMSIS\\5.5.1/CMSIS/DSP/Include" "-ID:\\Documents\\ArduinoData\\packages\\STM32\\hardware\\stm32\\1.9.0\\cores\\arduino" "-ID:\\Documents\\ArduinoData\\packages\\STM32\\hardware\\stm32\\1.9.0\\variants\\NUCLEO_L073RZ" "C:\\Users\\luke\\AppData\\Local\\Temp\\arduino_build_385493\\sketch\\stm32l073-control-prototype.ino.cpp" -o nul -DARDUINO_LIB_DISCOVERY_PHASE
Alternatives for max14871.h: [max14871]
ResolveLibrary(max14871.h)
  -> candidates: [max14871]
[... cut ...]
d:/documents/arduinodata/packages/stm32/tools/xpack-arm-none-eabi-gcc/9.2.1-1.1/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: stm32l073-control-prototype.ino.cpp:(.text.setup+0x6a): undefined reference to `LL_TIM_ENCODER_Init'
collect2.exe: error: ld returned 1 exit status
Using library max14871 in folder: D:\Documents\Arduino\libraries\max14871 (legacy)
Using library SrcWrapper at version 1.0.1 in folder: D:\Documents\ArduinoData\packages\STM32\hardware\stm32\1.9.0\libraries\SrcWrapper 
exit status 1
Error compiling for board Nucleo-64.

```

LukeGary462, 2021-03-16:
For those that run into the same issue, I resolved the problem by adding this in the platform.txt file:
```
compiler.extra_flags=-mcpu={build.mcu} {build.flags.fp} -mthumb "@{build.opt.path} "
```
Change that to this:
```
compiler.extra_flags=-mcpu={build.mcu} {build.flags.fp} -mthumb "@{build.opt.path} " -DUSE_FULL_LL_DRIVER
```

closed 2021-03-16 by LukeGary462

fpistm, 2021-03-16:
Hi @LukeGary462 
I've already defined the `USE_FULL_LL_DRIVER ` thanks #1321, this will be available with next release 2.0.0.
With current core you do not have to modify the platform.txt. You can use the build_opt.h to add some definition:
https://github.com/stm32duino/wiki/wiki/Customize-build-options-using-build_opt.h
by adding in the file:
`-DUSE_FULL_LL_DRIVER `