stm32duino/Arduino_Core_STM32 #1331: SPI2 hang on BlackPill F411

VitorBoss, 2021-03-20:
I'm having trouble using SPI2 on WeAct black pill F411, this is the SPI library example:
```
#include <SPI.h>
//            MOSI  MISO  SCLK
SPIClass SPI3(PC12, PC11, PC10);

void setup() {
  SPI3.begin(2); //Enables the SPI3 instance with default settings and attaches the CS pin  
  SPI3.beginTransaction(1, settings); //Attaches another CS pin and configure the SPI3 instance with other settings  
  SPI3.transfer(2, 0x52); //Transfers data to the first device
  SPI3.transfer(1, 0xA4); //Transfers data to the second device. The SPI3 instance is configured with the right settings  
  SPI3.end() //SPI3 instance is disabled
}
```
I can't send anything over SPI2 on the device, with a logic analyzer I can't see the traces

**Describe the bug**
When the SPI try to send a data it simply keep waiting for the flag `SPI_SR_TXE` got cleared and stuck at `spi_com.c` line 422

**To Reproduce**
Just run this
```
#include <SPI.h>
//            MOSI  MISO  SCLK
SPIClass spiToUse(PB15, PB14, PB13);

void setup() {
  spiToUse.begin(PB12);
  spiToUse.beginTransaction(PB12, SPISettings(1000000, MSBFIRST, SPI_MODE0));
  spiToUse.transfer(PB12, 0x06); //write enable
  spiToUse.transfer(PB12, 0x05); //read status register
  uint8_t answer = spiToUse.transfer(PB12, 0xFE); //dummy byte
  spiToUse.endTransaction(PB12);
  Serial.println(answer);
}
```

Steps to reproduce the behavior:
1. Upload the code
2. See error

**Expected behavior**
It should send the data or return an error after timeout

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Desktop (please complete the following information):**
 - OS: latest win10
 - Arduino IDE version: latest platforIO on VScode
 - STM32 core version: 1.9
 - Tools menu settings if not the default: build_flags = -g -std=gnu++11 -DARDUINO_BLACKPILL_F411CE -DHAL_UART_MODULE_ENABLED
 - Upload method: STLink

**Board (please complete the following information):**
 - Name: WeAct BlackPill F411
 - Hardware Revision: 1.3
 - Extra hardware used if any: FM25L256 FRAM chip but isn't needed for the error

ABOSTM, 2021-03-22:
HI @VitorBoss,
This is not a bug,
just that the SCLK pin you used is not the one defined for this board for this SPI instance:
When looking at PeripheralPins.c for BlackPill F411 (Generic_F411Cx)
the SPI2 SCLK signal is PB10.
PB13 is defined (reserved) for SPI4 SCLK (and thus commented out for SPI2).

So you can 
* either use PB10 for SPI2 SCLK
* or redefine in your sketch the `PinMap_SPI_SCLK[]` so as to define PB13 as SCLK for SPI2 (and comment out that pin for SPI4)
see https://github.com/stm32duino/wiki/wiki/Custom-definitions#custom-pinmap-array
```
const PinMap PinMap_SPI_SCLK[] = {
  {PA_5,  SPI1, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF5_SPI1)},
  {PB_0,  SPI5, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF6_SPI5)},
  //   {PB_3,  SPI1, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF5_SPI1)},
  {PB_3,  SPI3, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF6_SPI3)},
  {PB_10, SPI2, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF5_SPI2)},
  {PB_12, SPI3, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF7_SPI3)},
  {PB_13, SPI2, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF5_SPI2)},
  //{PB_13, SPI4, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF6_SPI4)},
  {NC,    NP,    0}
};
```

VitorBoss, 2021-03-22:
I'm following this image:
[image]

Here you can seed the PB12 been possible to use as NSS, also tried without library control and same result happens

VitorBoss, 2021-03-22:
[image]
From datasheet

ABOSTM, 2021-03-22:
1) What you show me is the hardware capabilities(independently from any software), what I told you about is how stm32duino core is currently implemented.
2) I don't say it is impossible, I just told it doesn't match the current blackpill F411 variant implementation.
and I provide you the way to use PB13 for SPI2 (redefining PinMap_SPI_SCLK[])
3) I don't understand why you are now talking about NSS and PB12 ... as far as I can read, you managed of NSS with PB12 is correct.

closed 2021-03-29 by fpistm