stm32duino/Arduino_Core_STM32 #1354: Loss of first output sent to Serial port (after firmware upload)

lnlp, 2021-04-06:
**Describe the bug**

After firmware upload via STLink, all output sent to the serial port within about the first second _(after call to Serial.begin())_ gets lost and does not show up in the serial monitor.

Adding a 'wait for serial ready' construct does not make any difference.
```cpp
Serial.begin(115200);
while (!Serial) 
{
}
``` 
Adding a delay of about 1.5 seconds after calling Serial.begin() 'solves' the symptoms but adding a delay is not preferred and  not a reliable solution.

```cpp
Serial.begin(115200);
delay(1500);
```

Arduino Core STM32 is the only core that I have experienced this issue with. I have used several other Arduino cores (e.g. AVR, SAMD, ESP32) and many other boards (either with MCU USB support, onboard USB to serial or no onboard USB). With none I experienced issues of missing the first output to the Serial monitor (both with Arduino IDE and PlotformIO). I have used the exact same USB to serial adapter and same sketch with a Pycom LoPy4 ESP32 board _(which also lacks onboard USB support)_. It has no problems with the first serial output, nothing gets lost.

I ran into this problem while writing a cross platform (boards/cores) example.

**To Reproduce**

I experience the problem with Bluepill and Blackpill (STMF103C8) boards. 
Firmware upload via STLink V2. Serial monitor connected using external USB to serial adapter connected to pins A9 and A10.
_(The boards do not use a bootloader with serial over USB support)._

Steps to reproduce the behavior:
1. Load below sketch in the Arduino IDE.
2. Select the Bluepill board in the Arduino IDE as target.
3. Connect a Bluepill board via STLink (for upload) and USB to serial adapter to pins A9, A10 (for Serial Monitor).
4. Select the proper USB to Serial COM port in the Arduino IDE.
5. Open the Serial Monitor.
6. Compile and upload below sketch.
7. Check the output. The first output should begin with "Printed after the delay". Everything printed before the delay will be missing.
8. Press the reset button.
9. Check the output again. All output will now be included.

The timing was added to see if it would provide any useful information (e.g. when comparing with other boards/cores) but it did not help in further pinpointing the cause of the problem.

```cpp
uint32_t start;
uint32_t after_begin;
uint32_t after_serial_check;
uint32_t before_first_output;
uint32_t after_first_output;
uint32_t after_delay;

void setup() 
{
    start = micros();
    
    Serial.begin(115200);
    after_begin = micros();
    
    while (!Serial)
    {
        delay(100);
    }
    after_serial_check = micros();

    before_first_output = micros();
    Serial.println("\nserial-output-loss-test");
    Serial.println("\nPrinted before the delay. This will probably get lost");
    after_first_output = micros();

    delay(1000);
    after_delay = micros();

    Serial.println("\nPrinted after the delay");    

    delay(2000);  // Just wait some more    Serial.println("This was printed after the delay);

    Serial.println("\nTiming in microseconds:\n");
    Serial.print("start:                ");
    Serial.println(start);
    Serial.print("after_begin:          ");
    Serial.println(after_begin);
    Serial.print("after_serial_check:   ");
    Serial.println(after_serial_check);
    Serial.print("before_first_output:  ");
    Serial.println(before_first_output);
    Serial.print("after_first_output:   ");
    Serial.println(after_first_output);
    Serial.print("after_delay:          ");
    Serial.println(after_delay);        

}

void loop() 
{

}
```

**Expected behavior**

Everything  that is printed to the serial port (after the call to `Serial.begin()`) should be visible in the Serial Monitor without any output missing.

**Screenshots**

Example output:

```txt
Bluepill + external USB to Serial adapter - output after upload:
----------------------------------------------------------------

Printed after the delay

Timing in microseconds:

start:                8
after_begin:          32
after_serial_check:   34
before_first_output:  35
after_first_output:   1936
after_delay:          1001002

Bluepill + external USB to Serial adapter - output after pressing the reset button:
-----------------------------------------------------------------------------------

serial-output-loss-test

Printed before the delay. This will probably get lost

Printed after the delay

Timing in microseconds:

start:                8
after_begin:          32
after_serial_check:   34
before_first_output:  35
after_first_output:   1936
after_delay:          1001002
```

**Desktop:**

 - OS: Windows 10 version 20H2
 - Arduino IDE version: 1.8.13
 - STM32 core version: 1.9.0
 - Tools menu settings if not the default: -
 - Upload method: STLink

**Board:**

 - Name: Bluepill 128k (STM32F103C8T6)
 - Hardware Revision: -
 - Extra hardware used: Wemos CH340G USB to serial adapter connected via Rx, Tx and GND.

**Additional context**

Similar problems when using PlatformIO (my primary IDE, I don't use the Arduino IDE).

ABOSTM, 2021-04-09:
Hi @lnlp,
I tried to reproduce and analyse the issue you describe.

**My setup:**
BluePill STM32F103C8T6 (standard 64K)
Upload with STLlink
USB to serial adapter: STLink different from the one used for upload (STLink V2 from a Nucleo board).

**Test result:**
With Arduino IDE, I could not reproduce the issue, everything is displayed on serial monitor
The same with other serial applications (TeraTerm and VScode)

**Nevertheless ...**
Nevertheless I succeeded to reproduce the behavior you described under very special conditions:
When using the same STlink (from a nucleo board) for both upload and serial monitor,
when I upload a sketch with Arduino IDE the 1st time after plugging USB,
then serial monitor display only "Printed after the delay ..."
But:
1) any further upload (without unplug of STlink) is well working
2) If I use a different serial application (teraTerm or VScode), I don't have problem
3) When behavior is reproduced, it looks like everything is normal from hardware point of view:
The logical analyser show me that all bytes are properly send on serial bus 
see screenshot below on the right side the decoded part 
```
 |   serial
 | -output-
 | loss-tes
 | t...Prin
 | ted befo
 | re the d
 | elay     
...
```
[image]

As you can see on the screenshot, there is an idle frame at HIGH level at the beginning. Duration is 86,6µs which correspond to a full frame (start + 8 data + Stop  with baudrate @115200).
This idle frame cannot be confused with a regular frame which have necessarily a rising edge between start and stop.
This idle frame is used to avoid any confusion and prevent misinterpretation of the initial rising edge which correspond to Uart initialization (Serial.begin). In other word after the idle frame the next falling edge is necessarily a start bit and this allow to synchronize transmitter with the receiver. So everything is there to have a proper transmitission. 
My guess is that Arduino IDE is not able to properly detect this idle frame and thus miss the first bytes.

Adding delay will artificially maintain the High state. So from hardware point of view it will simply extended the duration of the idle frame.
And I figure out  that a delay around 300ms is sufficient so that Arduino IDE serial monitor is able to display all the bytes.

fpistm, 2021-04-14:
Hi @lnlp 
any feedback on this else I will close it?
Thanks

lnlp, 2021-04-14:
I am about to do some additional test. Please leave open.

fpistm, 2021-04-22:
Closed as no feedback since a week.

closed 2021-04-22 by fpistm