stm32duino/Arduino_Core_STM32 #1355: Wire / I2C doesn't work on STM32F4

andrey-gvrd, 2021-04-12:
**Describe the bug**
The Wire library doesn't work. 

I'm monitoring the SDA, SCL pins with a logic analyzer and there aren't even any transitions. Both stay pulled up by the pull-up resistors.

Same result with both 1.8 and 1.9 versions of the core.

Possibly related to: https://github.com/stm32duino/Arduino_Core_STM32/issues/463

**To Reproduce**
I've tried the following combinations:

> Wire, I2C1: (PB7, PB6), (PB9, PB8), (PA10, PA9)
> TwoWire, I2C2: (PB11, PB10), (PB14, PB13)

Using the following code to test the library:
```
#include <Wire.h>

TwoWire Wire2(PB14, PB13);

HardwareSerial HWSerial2(PA3, PA2);

void setup() {
  HWSerial2.begin(115200);
  //Wire.setSDA(PA10);
  //Wire.setSCL(PA9);
  //Wire.begin();
  Wire2.begin();
  HWSerial2.println("\nI2C Scanner");
}

void loop() {
  byte error, address;
  int nDevices;

  HWSerial2.println("Scanning...");

  nDevices = 0;
  for(address = 1; address < 127; address++) {
    Wire2.beginTransmission(address);
    error = Wire2.endTransmission();
    
    if (error == 0) {
      HWSerial2.print("I2C device found at address 0x");
      if (address < 16) 
        HWSerial2.print("0");
      HWSerial2.println(address, HEX);

      nDevices++;
    }
    else if (error == 4) {
      HWSerial2.print("Unknown error at address 0x");
      if (address < 16) 
        HWSerial2.print("0");
      HWSerial2.println(address, HEX);
    }    
  }
  if (nDevices == 0)
    HWSerial2.println("No I2C devices found");
  else
    HWSerial2.println("done");

  delay(1000);
}
```

**Desktop:**
 - OS: Windows 10
 - Arduino IDE version: 1.8.13
 - STM32 core version: 1.9
 - Upload method: STLink (STM32CubeProgrammer SWD)

**Board:**
 - Name: STM32F40G-DISC1

**Additional context**
Using VL53L3CX board from Pololu for the test, but I don't think it's important.

andrey-gvrd, 2021-04-12:
Switched to PlatformIO to debug this further.
According to `PeripheralPins.c`, these are the valid I2C pins for this board:
```
WEAK const PinMap PinMap_I2C_SDA[] = {
	{PB_7,  I2C1, STM_PIN_DATA(STM_MODE_AF_OD, GPIO_NOPULL, GPIO_AF4_I2C1)},
	//  {PB_9,  I2C1, STM_PIN_DATA(STM_MODE_AF_OD, GPIO_NOPULL, GPIO_AF4_I2C1)}, // Audio_SDA
	{PB_11, I2C2, STM_PIN_DATA(STM_MODE_AF_OD, GPIO_NOPULL, GPIO_AF4_I2C2)},
	{PC_9,  I2C3, STM_PIN_DATA(STM_MODE_AF_OD, GPIO_NOPULL, GPIO_AF4_I2C3)},
	{NC,    NP,    0}
};

WEAK const PinMap PinMap_I2C_SCL[] = {
	{PA_8,  I2C3, STM_PIN_DATA(STM_MODE_AF_OD, GPIO_NOPULL, GPIO_AF4_I2C3)},
	//  {PB_6,  I2C1, STM_PIN_DATA(STM_MODE_AF_OD, GPIO_NOPULL, GPIO_AF4_I2C1)}, // Audio_SCL
	{PB_8,  I2C1, STM_PIN_DATA(STM_MODE_AF_OD, GPIO_NOPULL, GPIO_AF4_I2C1)},
	//  {PB_10, I2C2, STM_PIN_DATA(STM_MODE_AF_OD, GPIO_NOPULL, GPIO_AF4_I2C2)}, //  Mic CLK
	{NC,    NP,    0}
};
```

So (PB7, PB8) is the only pin combination that could work, but it still doesn't.

closed 2021-04-12 by andrey-gvrd

fpistm, 2021-04-12:
You can redefine the I2.C array