stm32duino/Arduino_Core_STM32 #1361: SPI.beginTransaction() not recognised

Laiaborrell, 2021-04-20:
Hi everyone,
the SPI library seems to work fine except when using the SPI.beginTransaction() function. It does not recognize it and shows the following error: "spi does not name a type". 
Can anybody help? 
Thankyou!!

fpistm, 2021-04-20:
Hi @Laiaborrell 
could you be more precise ?
Which Arduino IDE version? core version ? Boards? Sketch?

`SPI.beginTransaction()` is used in several sketch so it is very strange you got this error.

Laiaborrell, 2021-04-20:
Btw, sorry I am not giving any other information, I am kind of new in arduino and SPI.

fpistm, 2021-04-20:
Without further information we could not help you.

Laiaborrell, 2021-04-20:
Hi! Sorry, I am kind of new to Arduino and SPI. Arduino version is 1.8.13.  I am working with "NUCLEO-F401RE". The sketch is unfinished yet. However I need the beginTransaction function.

fpistm, 2021-04-20:
Here is the sketch used for CI build, as you can see the function is used and CI build fine.
https://github.com/stm32duino/Arduino_Core_STM32/blob/6e53b66040a7116242b10e54f3a89f5f4bbec72c/CI/build/examples/BareMinimum/BareMinimum.ino#L120

Laiaborrell, 2021-04-20:
I do't see why isn't it working then.

fpistm, 2021-04-20:
No not library version issue.
Maybe you do not use this core.
Could you copy a screenshot here of the Arduino Tools menu with the selected board, please?

Laiaborrell, 2021-04-20:
[image]
It is in catalan, but look at "Placa"

fpistm, 2021-04-20:
OK so it seems this is this core. Now which version of the core ? 1.9.0 ?
And also provide your sketch eve if not finished.

Laiaborrell, 2021-04-20:
What do you mean by core?

Laiaborrell, 2021-04-20:
How can I check my core version? Sorry I've been using arduino only for 2 months now.

fpistm, 2021-04-20:
This repository is the STM32 core to allow support of STM32 based boards.
You can read this for further information: https://www.arduino.cc/en/guide/cores

You have installed yourself the core thanks the Boards manager menu so you should be able to find it again.
Anyway provide your sketch because my guess is you simply define a variable with "spi" type and that's why you get this error:

> error: "spi does not name a type"

Laiaborrell, 2021-04-20:
[image]

fpistm, 2021-04-20:
Pleas copy the full sketch as text else I will close this issue.

Laiaborrell, 2021-04-20:
```
#include <SPI.h>

#define BME_SCK 13
#define BME_MISO 12
#define BME_CS  10
#define BME_MOSI 11
#define READ_TEMP 0x01
#define READ_PRESSURE 0x02

int tempVal=0;
int presVal=0;

//byte tempVal (definir la variable que emmagatzemarà el valor retornar per la funció readTemp)
//presVal (idem)

void setup() {
  // put your setup code here, to run once:  
  Serial.begin(115200); //configuramos la comunicación serie

  //Seteamos el pin CS como output
  pinMode(BME_CS, OUTPUT);
  pinMode(BME_MISO,INPUT);
  SPI.begin(); //inicializamos por hardware la SPI (no haría falta setear los pines MISO,MOSI y SCK)
}

void loop() {
  // put your main code here, to run repeatedly:
  SPI.beginTransaction(SPISettings(500000, MSBFIRST, SPI_MODE0));
  byte rxBuffer; //buffer recibido 
  if (Serial.available() > 0) { // si quedan bytes pendiente de leer: no sé si esto sirve en SPI
    rxBuffer = Serial.read(); //los leemos
  switch(rxBuffer){
    case READ_TEMP:
      tempVal=readTemp();
      Serial.println(tempVal); //we display the read value in the terminal
      break;
    case READ_PRESSURE:
      presVal=readPressure();
      Serial.println(presVal); //we display the read value in the terminal
      break;
    default: //si el comando enviado no corresponde con ninguno de los establecidos, avisamos con un 1
      Serial.println(0x01);    
      break;
   }
  }
}
```
I am not using the readPressure and readTemperature functions yet. However they are defined in this other sketch:

```
//Adresses dels registres del sensor:
const int MSB_TEMP = 0xFA; //most significant bytes temperature
const int LSB_TEMP = 0xFB; //least significant bytes temperature
const int MSB_PRES = 0xF7;
const int LSB_PRES = 0xF8;
byte msb_t=0;
int lsb_t=0;
int msb_p=0;
int lsb_p=0;

int readTemp(void) { //davant del nom de la funció he de posar el tipus de variable que retorna però no el sé perquè no sé definir la funció :) suposo q float
  digitalWrite(BME_CS, LOW);
  msb_t=SPI.transfer(MSB_TEMP|0x80); //10000000 (binary), estamos asegurandonos de que el byte 7 (0,1,...,7) es un 1, que es el comando de read
  lsb_t=SPI.transfer(LSB_TEMP|0x80);
  return msb_t;
}

int readPressure(void) {
  digitalWrite(BME_CS, LOW);
  msb_p=SPI.transfer(MSB_PRES|0x80);
  lsb_p=SPI.transfer(LSB_PRES|0x80);
  return msb_p;
}

```
Could this be the version?
[image]

fpistm, 2021-04-20:
No this is not the version... Simply open the boards manager in the tools -> boards submenu.

OK really not easy to understand your information.
Anyway, I think you have split your sketch in 2 files (2 tabs).

And the real error message is:
`error: 'SPI' does not name a type`
and not error: `"spi does not name a type"`

So this probably means you used `SPI` outside a function like here:
https://forum.arduino.cc/t/error-using-spi-begintransaction-spisettings-not-name-a-type/702397

I close this issue as if there is really an error on this function it will be raised since a while and we used daily it.

Try to build this simple sketch you will see it's build fine.

```c++
#include <SPI.h>

void setup() {
  SPI.beginTransaction(SPISettings(SPI_SPEED_CLOCK_DEFAULT, MSBFIRST, SPI_MODE_0));
}

void loop() {
}
```

closed 2021-04-20 by fpistm