stm32duino/Arduino_Core_STM32 #1362: analogRead(ATEMP) always returns 0 (zero) on STM32F411

AndrewBCN, 2021-04-21:
**analogRead(ATEMP) always returns 0 (zero) on STM32F411.** analogRead(AVREF) and analogRead(VBAT) return valid values.

Steps to reproduce the behavior:
Compile and run the example provided in the Wiki on any STM32F411 board.
https://github.com/stm32duino/wiki/wiki/API#example-2

**Expected behavior**
analogRead(ATEMP) should return some non-zero value which can then be used to calculate the internal temperature.

Desktop 
 - OS: Linux
 - Arduino IDE version: 1.8.13
 - STM32 core version: 1.9.0
 - Upload method: dfu

Board
 - Name: STM32F411CEU6 WeAct Black Pill
 - Hardware Revision: 3.0 (latest)
 - Extra hardware used if any: 

I have checked the code and the STM32F411 reference manual and I cannot find the problem. I am available to test and find the problem here, by compiling and running any test code (for example, to check that ADC channel 18 is correctly selected, etc).

On the STM32F411, ADC channel 18 is shared between VBAT and the temperature sensor, and by reading the code I can see that this is taken into account. But there must be something wrong somewhere, because AVBAT and AVREF both seem to read OK but ATEMP does not.

Also note that on the STM32F411CEU6, VBAT is fed to a 1/4 voltage divider before the ADC. This is not mentioned anywhere in the example in the Wiki, but it is documented in the datasheet.

fpistm, 2021-04-22:
Hi @AndrewBCN 

> Also note that on the STM32F411CEU6, VBAT is fed to a 1/4 voltage divider before the ADC. This is not mentioned anywhere in the example in the Wiki, but it is documented in the datasheet.

The example in the wiki is just provided as example and probably required some update depending of the boards.
Anyway, I've tested with a Nucleo F411RE and I have no issue:

```
VRef(mv)= 3313	Temp(°C)= 28	Vbat(mv)= 835	A0(mv)= 3312
VRef(mv)= 3315	Temp(°C)= 29	Vbat(mv)= 835	A0(mv)= 3314
VRef(mv)= 3315	Temp(°C)= 29	Vbat(mv)= 836	A0(mv)= 3312
VRef(mv)= 3315	Temp(°C)= 29	Vbat(mv)= 835	A0(mv)= 3315
```

Going deeper, in fact I've tested with 2.0.0 release and so it can explain why it works fine as the HAL/LL F4 has been updated.
Some new LL_ADC API have been added, ex: `__LL_ADC_CALC_VREFANALOG_VOLTAGE`
Like the Temp  displayed is computed using the vref read it can explains why it works.

Please could you test with the last core release, thanks.

AndrewBCN, 2021-04-22:
Thank you very much, Frédéric, and you are 100% right: I removed core release 1.9.0, installed core 2.0.0, recompiled, downloaded, and now the wiki example runs just perfect on the STM32F411CEU6 Black Pill board I am using. I am closing this issue.
Again, merci beaucoup!

closed 2021-04-22 by AndrewBCN

fpistm, 2021-04-22:
De rien! 😉