stm32duino/Arduino_Core_STM32 #1371: Code not working in v2.0.0

mrguen, 2021-04-26:
Hello,

After updating to 2.0.0 the classic code example from this library does compile but the TFT remains blank: [https://github.com/adafruit/Adafruit-ST7735-Library/tree/master/examples/graphicstest](https://github.com/adafruit/Adafruit-ST7735-Library/tree/master/examples/graphicstest)

I suppose this is related to the modifications on port names / addition of alternate functions. I can't tell more about this and rolling back to 1.9.0

fpistm, 2021-04-26:
@mrguen 
please provide more information. Which board? Which sketch?... all relevant information.
I've tested the TFT on several library without any issue...

fpistm, 2021-04-26:
@mrguen 
I close this issue as issue is not properly filled.
Title like: `Code not working in v2.0.0` is too generic.
No relevant information: board? IDE? GitHub template provides all information requested.
We expect from user to provide those information to be able to help.
As stated before it was tested on several boards.
Your supposition are wrong: port name has not been modified. No alternate functions has been added as they are related to CubeFW which is not changed here.
Like you can't tell more and rolling back to 1.9.0 no need to spend time on this.

closed 2021-04-26 by fpistm