stm32duino/Arduino_Core_STM32 #1377: Black Magic Probe with a two-digit COM port on Windows won't work

wirehead, 2021-04-30:
If there's a Black Magic Probe and you've run enough different serial port devices into your system such that it's a two-digit COM port, the Arduino integration doesn't properly prefix the path and thus the Arduino app is unable to connect to the Black Magic Probe.

**To Reproduce**

Steps to reproduce the behavior:
1. Plug enough different USB serial devices devices into your computer that COM1-9 are reserved.  (This is either a normal day in the life or a scavenger hunt, LOL)
2. Plug a Black Magic Probe in, which will now reserve two COM ports with two-digit port numbers.
3. Configure the the upload method to BMP, other settings don't really matter.
4. Try to upload and notice the error message when it can't connect to the COM port.

**Expected behavior**

Arduino should at least connect to the Black Magic Probe and get farther along on the flashing process.

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Arduino IDE version: 1.8.13
 - STM32 core version: 2.0.0
 - Upload method: BMP (Black Magic Probe)

**Board (please complete the following information):**
 - Name: [e.g. Nucleo F429ZI]
 - Hardware Revision: [e.g. Rev B]
 - Extra hardware used if any: [e.g. X-Nucleo IKS01A1]

**Additional context**

It's a very Windows-specific thing, brought because of a legacy that's older than STMicroelectronics, LOL.

As shown on https://github.com/blacksphere/blackmagic/wiki/Getting-Started notice the quote:

On Windows, use the the COM port instead of `/dev/ttyACM0`. For ports >= COM10, add the prefix `\\.\`, for example:
```
target extended-remote COM3
target extended-remote \\.\COM10
```

**Temporary workaround**

Go into the Device manager, select "Show hidden devices" and go to the "Ports" section, right-click on each one and select "Uninstall device" to delete all of your serial ports or at least enough so that COM8 and COM9 are free, then connect your BMP again.

fpistm, 2021-05-03:
Hi @wirehead 
from my point of view it is a restriction of the BMP.
On Arduino side upload BMP method is called properly with `"target extended-remote {serial.port}"` :
https://github.com/stm32duino/Arduino_Core_STM32/blob/0a3fa54b4a645eba9f9e5db9165a85c3bfc6d66e/platform.txt#L183

This is the same for all host supported OS. Why specifically check if COM port has 2 digits on Windows and change command lien?
This could not be achieve easily without adding extra batch script while it is a tools restrictions.
I guess the only thing we can do here is add a restriction note in the Wiki for this upload method. 
Moreover I don't think this one is used a lot as it was added for legacy with https://github.com/rogerclarkmelbourne/Arduino_STM32 core.
It also have the same restriction:
https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/bab4fda0fa31ba2d1f91f45329843d2df77da74b/STM32F1/platform.txt#L154

fpistm, 2021-05-03:
Warning has been added in the Wiki:
https://github.com/stm32duino/wiki/wiki/Upload-methods#bmp-black-magic-probe
https://github.com/stm32duino/wiki/wiki/FAQ#when-using-bmp-black-magic-probe-upload-method-error-occurs-it-cant-connect-to-the-com-port

reppad, 2022-06-07:
Hi @fpistm,
The syntax with the prefix `\\.\` seems to works also with port numbers lower than 10, so why not include it systematically in the upload command for windows by adding a upload.pattern.windows variant in platform.txt ?

`tools.bmp_upload.upload.pattern.windows="{path}/{cmd}" -nx -b {upload.speed} {upload.verbose} -ex "set confirm off" -ex "target extended-remote \\.\{serial.port}" -ex "monitor swdp_scan" -ex "attach 1" -ex "load" -ex "compare-sections" -ex "kill" "{build.path}/{build.project_name}.elf"`

I can do a PR if this way is ok for you ?

esden, 2022-06-07:
Just to clarify, this is not a limitation of BMP in any way. BMP has nothing to do with any of this. You are using GDB to connect to a remote target com port that is provided by BMP. The correct way to provide remote com port targets to GDB on Windows was pretty much always with the `\\.\` prefix. (At least since Windows NT kernel. The prefix is essentially windows kernel internals leaking into the userspace.)

fpistm, 2022-06-07:
Thanks for the feedback @esden.
Will fix that.

closed 2022-06-08 by fpistm