stm32duino/Arduino_Core_STM32 #1392: Use the SWD pins for Serial communication on Blackpill411

SensorsIot, 2021-05-12:
**Describe the bug**
I tried to use HardwareSerial(PA13,PA14) on a Blackpill 411 and it does not work. It works on other pins
**To Reproduce**
use a simple sketch that prints something

Steps to reproduce the behavior:
Standard upload via USB

**Expected behavior**
We want to use the SWD pins for Serial communication when not used for programming

**Desktop (please complete the following information):**
 - OS: Windows
 - Arduino IDE version: 1.8.13
 - STM32 core version: 2.0.0
 - Tools menu settings if not the default: Generic Serial/ Generic Serial supersede
 - Upload method: USB/DFU

**Board (please complete the following information):**
 - Name: Blackpill411CE
 - Hardware Revision: ?
 - Extra hardware used if any: none

fpistm, 2021-05-12:
Hi @SensorsIot 

PA13 and PA14 have no U(S)ART capabilities that's why you get nothing.

[image]

SensorsIot, 2021-05-12:
Thank you for your answer. Would it be possible to use these pins with softwareSerial?

fpistm, 2021-05-12:
You can try anyway I do not recommand it as it consumes lot of CPU but it depends on your use case.
If you can use an HardwareSerial you will save your time, I guess.

fpistm, 2021-05-12:
https://github.com/stm32duino/wiki/wiki/Libraries#built-in-delivered-with-the-core-package

> SoftwareSerial: follow the [official SoftwareSerial](https://www.arduino.cc/en/Reference/softwareSerial) API.
> It uses an hardware timer. Interrupt is triggered at a frequency = Baudrate * OVERSAMPLE (by default 3*baudrate) to handle RX as well as TX transmissions. This impacts CPU load specially at high baudrate.

SensorsIot, 2021-05-12:
Thank you. I tried it and so far TX worked on both pins. I assume RX will also work. Like that we can have a simple 3 wire (RX, TX, RST) interface for programming and communication. Cool. We are working on a very useful project with it...

closed 2021-05-12 by SensorsIot

fpistm, 2021-05-12:
Welcome! Thanks for the feedback.