stm32duino/Arduino_Core_STM32 #1394: Problems with serial ports on STM32F103R(C-D-E)T variant

pkourany, 2021-05-13:
OS: Win10 x64 (latest)
Arduino IDE version: 1.8.13
STM32 core version: 2.0.0
Tools menu settings if not the default: Newlib Standard, No Serial
Upload method: Stm32CubeProgrammer(SWD)
Board: STM32F1 Series

In porting my F103R(C-D-E)T variant from 1.9.0 to 2.0.0, I now have compile errors in regards to serial ports.  Besides minor changes (using HSI only (no PLL), no button, LED pin change), the variant uses USARTs 1, 2, 3 and UART4 with "Serial" specified as "Serial1" (USART1) on pins PA9, PA10.

In my code, I initialize Serial2 (USART2), Serial3 (USART3) and Serial4 (UART4) using the following code, leaving USART1 to be configured automatically as "Serial".
```
HardwareSerial Serial2(PA3, PA2);		// USART2
HardwareSerial Serial3(PB11,PB10);	// USART3
HardwareSerial Serial4(PC11,PC10);	// UART4
```
This has worked perfectly until v2.0.0 where I get a linking error:
`(HardwareSerial.cpp.o):(.bss.Serial2+0x0): multiple definition of `Serial2'`, stating it was first defined in my code above.

I have not had any success getting past this compile error except when I comment out the `HardwareSerial Serial2(PA3, PA2);		// USART2` line.  However, in testing, the Serial port is not outputting expected messages so something else is going on.  Any guidance is appreciated.

fpistm, 2021-05-13:
The generic defines by default Serial to be Serial2
https://github.com/stm32duino/Arduino_Core_STM32/blob/655587065fbdd6324655fee0b13d268507e23a40/variants/STM32F1xx/F103R(C-D-E)T/variant_generic.h#L174

It is not very clear what you mean by porting your variant and the No serial in the tools menu. You would tell no generic Serial instance?

pkourany, 2021-05-13:
Sorry, I goofed.  The correct Tools menu settings are:
USB support: None
generic Serial
Newlib Nano (default)

My variant defines:
```
#ifndef SERIAL_UART_INSTANCE
  #define SERIAL_UART_INSTANCE  1	//ex: 2 for Serial2 (USART2)
#endif

#ifndef PIN_SERIAL_RX
  #define PIN_SERIAL_RX         PA10
#endif
#ifndef PIN_SERIAL_TX
  #define PIN_SERIAL_TX         PA9
#endif

  #ifndef SERIAL_PORT_MONITOR
    #define SERIAL_PORT_MONITOR   Serial
  #endif
  #ifndef SERIAL_PORT_HARDWARE
    #define SERIAL_PORT_HARDWARE  Serial1
  #endif
```

fpistm, 2021-05-13:
OK for the menu but for the variant how did you define it ?

pkourany, 2021-05-13:
What do you need to see?  I created `PeripheralPins_xxx.c` and `variant_xxx.cpp and .h` files (see attached)
[variant.zip](https://github.com/stm32duino/Arduino_Core_STM32/files/6473864/variant.zip)

fpistm, 2021-05-13:
OK so I think the issue are in the boards.txt you have to define the variant_h recipe else the generic one is used.

You should have something like this:

```
GenF1.menu.pnum.ANNEXIA_F103RCT=ANNEXIA F103RCT
GenF1.menu.pnum.ANNEXIA_F103RCT.upload.maximum_size=524288
GenF1.menu.pnum.ANNEXIA_F103RCT.upload.maximum_data_size=65536
GenF1.menu.pnum.ANNEXIA_F103RCT.build.board=ANNEXIA_F103RCT
GenF1.menu.pnum.ANNEXIA_F103RCT.build.product_line=STM32F103xE
GenF1.menu.pnum.ANNEXIA_F103RCT.build.variant=STM32F1xx/F103R(C-D-E)T
GenF1.menu.pnum.ANNEXIA_F103RCT.build.variant_h=variant_{build.board}.h
GenF1.menu.pnum.ANNEXIA_F103RCT.build.peripheral_pins=-DCUSTOM_PERIPHERAL_PINS
```

pkourany, 2021-05-13:
Thanks!  I had all but the custom peripherals line.  I'll test and report back.

UPDATE: Besides the added line, it turns out that I must have deleted the "underscore" in `build.variant_h` which likely caused the default file to be used.  Now I get zero errors.  Thanks Frederic!

closed 2021-05-13 by pkourany