stm32duino/Arduino_Core_STM32 #1395: STM32L(RAK811 chip) ADC value=0 in arduino IDE

bioshock2k, 2021-05-17:
Hi,

I´ve been trying to use the RAK 811 chip ADC that runs a STM32L. When using the Arduino IDE using analogRead() i get a value of 0 in the read.

AM i missing something?

fpistm, 2021-05-17:
Hi @bioshock2k 
could you be more precise?
Arduino IDE version? Core version?
Board selected? (there are 2 RAK 811)? Menu options?
Which pin you used with `analogRead()`?
...
All relevant information else we could not help.

bioshock2k, 2021-05-17:
Yes i can.
Arduino IDE 1.8.49
Board selected: LoRa boards
Board Part Number: RAK811 Lora tracker(16kb RAM)
All these settings i had sucess programming and messing with the RAK811, but never with the adc.

This is the code i am using to test the ADC:

`#include <arduino_lmic.h>
#include <arduino_lmic_hal_boards.h>
#include <arduino_lmic_hal_configuration.h>
#include <arduino_lmic_lorawan_compliance.h>
#include <arduino_lmic_user_configuration.h>
#include <lmic.h>

#include "STM32LowPower.h"
#include <lmic.h>
#include <hal/hal.h>
#include <SPI.h>

#include <Arduino.h>
#include <NewPing.h>
#include <MedianFilter.h>
#include <Wire.h>

int ADC_pin=PA3;
int ADC_pin2=PB14;
// Pin mapping
const lmic_pinmap lmic_pins = {
    .nss = 26,
    .rxtx = 32,
    .rst = 21,
    .dio = {27, 28, 29},
};

void setup() {
  // put your setup code here, to run once:
    Serial.begin(115200);
    while ( !Serial ) delay(10);
    Serial.println("\r\n*******************************");
    Serial.println("System Starting...");
    Serial.println("");
    pinMode(ADC_pin, INPUT);
    pinMode(ADC_pin2,OUTPUT);
    //os_init();
}

void loop() {
  // put your main code here, to run repeatedly:
  float battery=analogRead(ADC_pin);
 // int battery2=digitalWrite(ADC_pin2,HIGH);
  Serial.print("Batery: ");
  Serial.println(battery);
  //Serial.println(battery2);
  delay(500);
}`

fpistm, 2021-05-17:
Thanks
Pins used are not defined by the variant as ADC pins:

> ```c++
> int ADC_pin=PA3;
> int ADC_pin2=PB14;
> ```

Only those ones have been defined:
https://github.com/stm32duino/Arduino_Core_STM32/blob/655587065fbdd6324655fee0b13d268507e23a40/variants/STM32L1xx/L100C6Ux(A)_L151C(6-8-B)(T-U)x(A)_L152C(6-8-B)(T-U)x(A)/variant_RAK811_TRACKER.h#L34-L36

Do not ask me why.  This is a contribution and I never had this board. 😉 So I don't know why contributor does not defined them. Anyway that's why you get always 0 as it is not considered as a valid analog pin.

bioshock2k, 2021-05-17:
Thank you very much!
That will do perfectly. 
Could i add those ADC pins as a contributor?

fpistm, 2021-05-17:
> Could i add those ADC pins as a contributor?

Of course. Any contributions are welcome.

fpistm, 2021-05-24:
Hi @bioshock2k 
Please, have you the list of ADC pin to add ?

bioshock2k, 2021-05-24:
The list of available ADC pins on the RAK811?

fpistm, 2021-05-24:
> 
> 
> The list of available ADC pins on the RAK811?

yes 😉 The one which should be exposed.

bioshock2k, 2021-05-24:
2 | PB12/ADC | I/O | GPIO and ADC
3 | PB14/ADC | I/O | GPIO and ADC
4 | PB15/ADC | I/O | GPIO and ADC
20 | PA2/ADC | I/O | GPIO and ADC
22 | PA1/ADC | I/O | GPIO and ADC
23 | PA0/ADC | I/O | GPIO and ADC

Here it goes

fpistm, 2021-05-24:
OK, I guess this is the one defined here and so you used the [RAK811 Wis DUO](https://store.rakwireless.com/products/rak811-lpwan-module)
https://docs.rakwireless.com/Product-Categories/WisDuo/RAK811-Module/Datasheet/#low-rf-version-pin-definition

[image]

but in the core this is the [RAK811 LoRa Tracker](https://www.thethingsnetwork.org/marketplace/product/rak811-lora-tracker) (which seems no more sold)
[image]

So there is no issue in the variant. The [RAK811 LoRa Tracker](https://www.thethingsnetwork.org/marketplace/product/rak811-lora-tracker) has only 3 pins available for ADC (PA0 is connected to measure VBAT)

as the pin mapping is not the same a new variant should be created.

fpistm, 2021-05-25:
Added to the [New Variant] request list #722

closed 2021-05-25 by fpistm