stm32duino/Arduino_Core_STM32 #1409: cannot import name 'IS_WINDOWS' from 'platformio.compat'

danidask, 2021-06-03:
trying to build, it gives me this error. The code is a basic blink example

**Desktop**
 - OS: Windows 10 64bits
 - Platformio Core 5.1.1·Home 3.3.4
 - Platform ST STM32 14.0.0

platform.ini
```
[env:nucleo_f103rb]
platform = ststm32
board = nucleo_f103rb
framework = arduino
```

Output
```
Processing nucleo_f103rb (platform: ststm32; board: nucleo_f103rb; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/nucleo_f103rb.html
PLATFORM: ST STM32 (14.0.0) > ST Nucleo F103RB
HARDWARE: STM32F103RBT6 72MHz, 20KB RAM, 128KB Flash
DEBUG: Current (stlink) On-board (stlink) External (blackmagic, cmsis-dap, jlink)
PACKAGES:
 - framework-arduinoststm32 4.20000.210601 (2.0.0)
 - framework-cmsis 2.50700.210515 (5.7.0)
 - toolchain-gccarmnoneeabi 1.90201.191206 (9.2.1)
ImportError: cannot import name 'IS_WINDOWS' from 'platformio.compat' (c:\users\user\.platformio\penv\lib\site-packages\platformio\compat.py):
  File "C:\Users\user\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 177:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\user\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Script\SConscript.py", line 591:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\user\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Script\SConscript.py", line 280:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\user\.platformio\platforms\ststm32\builder\main.py", line 120:
    target_elf = env.BuildProgram()
  File "C:\Users\user\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Util.py", line 658:
    return self.method(*nargs, **kwargs)
  File "C:\Users\user\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 61:
    env.ProcessProgramDeps()
  File "C:\Users\user\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Util.py", line 658:
    return self.method(*nargs, **kwargs)
  File "C:\Users\user\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 127:
    env.BuildFrameworks(env.get("PIOFRAMEWORK"))
  File "C:\Users\user\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Util.py", line 658:
    return self.method(*nargs, **kwargs)
  File "C:\Users\user\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 342:
    SConscript(env.GetFrameworkScript(f), exports="env")
  File "C:\Users\user\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Script\SConscript.py", line 654:
    return method(*args, **kw)
  File "C:\Users\user\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Script\SConscript.py", line 591:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\user\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Script\SConscript.py", line 280:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\user\.platformio\platforms\ststm32\builder\frameworks\arduino.py", line 50:
    SConscript(build_script)
  File "C:\Users\user\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Script\SConscript.py", line 654:
    return method(*args, **kw)
  File "C:\Users\user\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Script\SConscript.py", line 591:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\user\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Script\SConscript.py", line 280:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\user\.platformio\packages\framework-arduinoststm32\tools\platformio\platformio-build.py", line 28:
    from platformio.compat import IS_WINDOWS
=================================================================== [FAILED] Took 1.22 seconds ==================================
```

fpistm, 2021-06-03:
Hi @danidask 
we support only Arduino IDE not PIO.
You should submit an issue here:
https://github.com/platformio/platform-ststm32

So I close this one.

closed 2021-06-03 by fpistm

fpistm, 2021-06-03:
For ref https://github.com/platformio/platform-ststm32/issues/529