stm32duino/Arduino_Core_STM32 #1412: stm32duino v2.0.0

fpgiovanini, 2021-06-05:
Dear Stm32duino 
I´m having the following troubles in stm32duino v2.0.0:
In version 2.0.0 (With the same arduino sketch ) I´m having the erros message "rodata' will not fit in region `FLASH'"
In version 1.9.0 this error don´t appear.
I couldn´t reverse to use the version 1.9.0 using the arduino board manager URL even the git procedure.
The board stm32f030c8t6  the  USART1 (Serial.print instruction ) don´t work.
tks, best regards

fpistm, 2021-06-05:
Dear @fpgiovanini 
there is no issue. The 2.0.0 now provide all pins possibilities in the peripheralPins.c (not in the 1.9.0). This consume more flash.
That's why the .rodata section not fit in region Flash. You can reduce the size by redefining the pinmap array according to your needs:
https://github.com/stm32duino/wiki/wiki/Custom-definitions#custom-pinmap-array

You could go back to the 1.9.0 using the old json link. (you can even have 1.9.0 installed and the 2.0.0 at the same time as they are installed in different path due to maintainer name change (STM32 to STMicroelectonics):
New json (2.0.0); https://github.com/stm32duino/BoardManagerFiles/raw/master/package_stmicroelectronics_index.json
Old json (<= 1.9.0); https://github.com/stm32duino/BoardManagerFiles/raw/master/STM32/package_stm_index.json

About the Usart1, it you talk about the generic stm32f030c8 then it defines the Serial to used USART2:
https://github.com/stm32duino/Arduino_Core_STM32/blob/28415c1bef214068a98ebec92b6014edd3689fbb/variants/STM32F0xx/F030C8T/variant_generic.h#L123-L135

closed 2021-06-05 by fpistm

fpgiovanini, 2021-06-05:
Dear fpistm
Sorry for my lack of ability.Thank you very much, best regards!

fpistm, 2021-06-05:
No worry. Lot of changes not easy to follow all changes  😉

fpgiovanini, 2022-01-14:
Dear Frederic

First of all, happy new year!!!!
Frederic I was guessing if you could help me with the following issue:
I need to use two SPI ports in a STM32F401. (SPI1 (PA7, PA6, PA5)😉 and SPI2 (PB15, PB14, PB13)  MOSI  MISO  SCLK
pins repectively.

I declared SPIClass SPI_2(PB15, PB14, PB13); in my sketch and it works, but SPI1 don´t work and craches the processor.

SPI.begin();
SPI_2.begin(); //Initiallize the SPI 2
SPI_2.beginTransaction(SPI_SETTINGS)

Could you please send me any link with this issue treatment?
Thanks a lot, best regards.

fernando

Enviado do Outlook<http://aka.ms/weboutlook>

https://github.com/stm32duino/wiki/wiki/Custom-definitions#custom-pinmap-array

https://github.com/stm32duino/Arduino_Core_STM32/blob/28415c1bef214068a98ebec92b6014edd3689fbb/variants/STM32F0xx/F030C8T/variant_generic.h#L123-L135

ABOSTM, 2022-01-17:
Hi @fpgiovanini,
Happy new year !!!
Can you please open a new issue, as this one is already closed.
Did you try to use SPI2 only ? does it work ?
Also can you check each of your SPI device on SPI1 in stand alone (without other SPI device),
to make sure this is not the device which causes issue?

fpgiovanini, 2022-01-17:
Hi Alexander.
I´m glad to hear from you.Thanks.

Did you try to use SPI2 only ? A: yes
does it work ?A: yes

Also can you check each of your SPI device on SPI1 in stand alone (without other SPI device),
to make sure this is not the device which causes issue?
A: the device works well on SPI1 of other uProcessor.
I atacched the schematic to you have an idea of the project.
ESP32 two SPI´s ports --work well
STM32F401 only SPI2 works -SPI1 CRACHES THE PROCESSOR.

Thanks a lot, best regards!!

fernando

Enviado do Outlook<http://aka.ms/weboutlook>