stm32duino/Arduino_Core_STM32 #1418: [U(S)ART] request list

fpistm, 2021-06-11:
This issue aims to list all U(S)ART requests:

- [ ] Hardware Flow control for serial communication (#394)
- [ ] Support for UART RX/TX level inversion on F3/F7 platforms (#1160)
- [ ] HardwareSerial Error Callback( #1162)

## Note

**This is a community projects,  Any contribution is welcome, so feel free submit a PR.** 😉

4ndreas, 2021-07-06:
I needed the 9 Bit Mode for a project and changed all the buffers to uint16_t and it worked. But I'm not sure if there are some side effects.
Could also solved with a flag for the 9th bit.

fpistm, 2021-07-06:
Hi @4ndreas 
THis was discussed some years ago:
https://github.com/stm32duino/Arduino_Core_STM32/issues/482#issuecomment-476229801

Main risk is the compatibility issue.

matthijskooijman, 2021-08-05:
I have one more suggestion for the list: Support serial writes when interrupts are disabled (e.g. from an ISR). This would make debugging a lot easier, and prevent surprise when someone prints from an ISR and the entire board locks up as soon as the buffer is full.

To support this usecase, the busy-wait loops in `flush()` and `write()` must be modified to call the ISR (or other code that moves the buffer into the UART) whenever a bit is written.

I implemented something similar on the Arduino AVR core, where IIRC it checks if interrupts are disabled, and if so, checks the TXE bit inside the busywait loop and calls the ISR if it is set. A similar approach can be taken here, though there are some complications:
 - There is not just a single "interrupts disabled" bit, AFAIU when another interrupt is running, the global interrupt mask is unchanged, but IRQ priorities might still prevent the TX interrupt from firing. To simplify things, I think the busy wait loops should simply disable the TXE/TXC (or whichever is used exactly) before the loop, and then inside the loop *always* call the ISR manually, even when interrupts were enabled already.
 -  There is some additional layering with the HAL, the stm32/uart.c and HardwareSerial that makes it a bit more tricky to get this right.

The second point makes this more complicated than I have time for right now, but I at least wanted to share this request and my thoughts.

fpistm, 2021-08-10:
Thanks @matthijskooijman for sharing.
Just FYI it exists an uart debug feature which does not rely on IT: `serial_debug`
Defining DEBUG_UART (USART peripheral, ex: USART1) and `DEBUG_PINNAME_TX` (PY_n) allows to specify the pin to use else by default it uses the same than default Serial instance. This fallback to `uart_debug_write`.

akasona, 2021-08-20:
> This is a community projects, Any contribution is welcome, so feel free submit a PR.

@fpistm I want to make PR for the item: **Hardware Flow control for serial communication**.
Where can I find any rules or workflows to contribute this project?

fpistm, 2021-09-07:
> 
> 
> > This is a community projects, Any contribution is welcome, so feel free submit a PR.
> 
> @fpistm I want to make PR for the item: **Hardware Flow control for serial communication**.
> Where can I find any rules or workflows to contribute this project?

Hi @akasona 
Sorry, I missed your comment. 
You can refers to the https://github.com/stm32duino/Arduino_Core_STM32/blob/master/CONTRIBUTING.md
To resume:
- Fork
- Implement
- Ensure Astyle is passed
- Submit the PR
- That's all folks 😉 

-

fpistm, 2021-12-16:
Hi @ht93 , it seems not available fo F4.

ht93, 2021-12-16:
Hi @fpistm, is there any other way to implement this? I am current working on a design with very limit size. Thanks a lot!

akasona, 2022-01-26:
@fpistm I'm sorry for so slow action! Finally I made up PR in #1634.

akasona, 2022-06-22:
Added flow control APIs
https://github.com/stm32duino/wiki/wiki/API#enable-hardware-flow-control

anatom74, 2022-11-09:
a suggestion to support for swapping RX/TX pins 
(Bit 15 SWAP: Swap TX/RX pins)

mhmayyan, 2024-10-29:
Hi
I am unable to start UART with a 1200 baud rate. I used Serial.begin(1200), but no data were transmitted. When I try a 2400 and higher baud rates, it works. Does anyone have any idea?

fronders, 2024-10-30:
@mhmayyan It's probably that the BRR value would overflow 0xFFFF for your desired rate at given core frequency.
IIRC the core runs by default at 80MHz so that would give you bauds between 1220 bps and 5MBps. I don't know if you check the return code, but at 1200 it would probably just not initialize and return an error

[image]

p.s. I'd recommend opening a separate issue not to flood the request list

fpistm, 2024-10-30:
Hi @mhmayyan 
@fronders gave the right answer.
BRR is overflowed, you have to decrease the clock config and for this kind of question you can open a GitHub discussion.

closed 2025-03-31 by fpistm