stm32duino/Arduino_Core_STM32 #1419: ST7735 Display Problems, Pins not working correctly, STM32F103RB

ColdFireHunter, 2021-06-11:
Hi,

**Describe the bug**
When I connect an ST7735 LCD Module to my custom STM32F103RB Board the Display doesn't show anything. 
In a few very rare cases it does show the right program. (after long power off or cycling the power 10 to 20 times)

**To Reproduce**
Use the Adafruit ST7735 LCD Libary with the Graphics Test sketch.
Connections:
SCLK --> PA5
MOSI --> PA7   //NORMAL SPI 1
CS --> PA4
DC --> PA2
RST --> PA3
GND --> GND
VDD --> 3.3V

Standard Clock Configuration with HSI. Core Clock: 64MHz

Steps to reproduce the behavior:
1. Upload Sketch
2. Watch Display

**Expected behavior**
Graphics Test Pattern on the screen

**Screenshots**

**Desktop:**
 - OS: Windows
 - PlatformIO V.5.1.1
 - STM32 core version: 2.0.0
 - Tools menu settings if not the default: Default
 - Upload method: SWD

**Board:**
 - Name: STM32F103RB
 - Hardware Revision: DIY Hardware
 - Extra hardware used if any: ST7735 Display 1.8" 160x128 pixels

**Additional context**
I already checked all my connections with a Multimeter there are fine.
When I use different pins (RST, DC, CS) on an F103RB Nucleo (Digital 8,9,10), the Display works just fine.
I also tried 4 other displays with the same Controller and none worked with these pins.
In Core Version 1.9.0 everything worked fine.

fpistm, 2021-06-11:
Hi @ColdFireHunter 
Which variant you used and please test with Arduino IDE. PIO is not supported here.

fpistm, 2021-06-11:
Looking at your pin mapping, PA2 and PA3 are used for the default Serial Instance for both F103RB target (generic and nucleo)

For Nucleo F103RB:
https://github.com/stm32duino/Arduino_Core_STM32/blob/c20257076fe0cde80561026c7e3aedaefd4bc95b/variants/STM32F1xx/F103R(8-B)T/variant_NUCLEO_F103RB.h#L125-L130

For  Generic F103RB:
https://github.com/stm32duino/Arduino_Core_STM32/blob/c20257076fe0cde80561026c7e3aedaefd4bc95b/variants/STM32F1xx/F103R(8-B)T/variant_generic.h#L160-L165

On 1.9.0, the generic F103RB  defines other pins that's why it works: 
https://github.com/stm32duino/Arduino_Core_STM32/blob/17fd01f9ff569cc55de975da6a5742a8aab6d004/variants/Generic_F103Rx/variant.h#L122-L124

closed 2021-06-11 by fpistm