stm32duino/Arduino_Core_STM32 #1443: Buffered EEPROM functions are not compling in new core 2.0.0

Vicky-S, 2021-07-13:
**Desktop (please complete the following information):**
 - OS: [e.g. Windows]
 - Arduino IDE version: [1.8.15]
 - STM32 core version: [ 2.0.0]
 - Tools menu settings if not the default: [e.g. Newlib Standard, No Serial]
 - Upload method: [e.g. SWD]

Board Name: STM32F411RE

The program was done in 1.9.0 and working well with that core but it fails to compile in the new core. The arduino gives the following error messages.

"EEPROM:79:3: error: 'eeprom_buffer_flush' was not declared in this scope
   79 |   eeprom_buffer_flush();
      |   ^~~~~~~~~~~~~~~~~~~
C:\Users\Admin\Downloads\WIM_5_Series_MCL_V1.4.4_9600 (2)\WIM_5_Series_MCL_V1.4.4_9600 (1)\WIM_5_Series_MCL_V1.4.4.4_Senlogic_Protocol\EEPROM.ino: In function 'void EEPROM_read()':
EEPROM:86:3: error: 'eeprom_buffer_fill' was not declared in this scope
   86 |   eeprom_buffer_fill();
      |   ^~~~~~~~~~~~~~~~~~
C:\Users\Admin\Downloads\WIM_5_Series_MCL_V1.4.4_9600 (2)\WIM_5_Series_MCL_V1.4.4_9600 (1)\WIM_5_Series_MCL_V1.4.4.4_Senlogic_Protocol\EEPROM.ino: In function 'void EEPROMWritelong(int, long int)':
EEPROM:163:3: error: 'eeprom_buffered_write_byte' was not declared in this scope
  163 |   eeprom_buffered_write_byte(address, four); delay(1);
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~
C:\Users\Admin\Downloads\WIM_5_Series_MCL_V1.4.4_9600 (2)\WIM_5_Series_MCL_V1.4.4_9600 (1)\WIM_5_Series_MCL_V1.4.4.4_Senlogic_Protocol\EEPROM.ino: In function 'void EEPROMWritefloat(long int, float)':
EEPROM:177:3: error: 'eeprom_buffered_write_byte' was not declared in this scope
  177 |   eeprom_buffered_write_byte(address, four); delay(1);
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~
C:\Users\Admin\Downloads\WIM_5_Series_MCL_V1.4.4_9600 (2)\WIM_5_Series_MCL_V1.4.4_9600 (1)\WIM_5_Series_MCL_V1.4.4.4_Senlogic_Protocol\EEPROM.ino: In function 'long int EEPROMReadlong(long int)':
EEPROM:185:16: error: 'eeprom_buffered_read_byte' was not declared in this scope
  185 |   byte four  = eeprom_buffered_read_byte(address);
      |                ^~~~~~~~~~~~~~~~~~~~~~~~~
C:\Users\Admin\Downloads\WIM_5_Series_MCL_V1.4.4_9600 (2)\WIM_5_Series_MCL_V1.4.4_9600 (1)\WIM_5_Series_MCL_V1.4.4.4_Senlogic_Protocol\EEPROM.ino: In function 'float EEPROMReadfloat(long int)':
EEPROM:195:16: error: 'eeprom_buffered_read_byte' was not declared in this scope
  195 |   long four  = eeprom_buffered_read_byte(address);"

fpistm, 2021-07-13:
Please provide the sketch and the selected board not only the mcu name.

Vicky-S, 2021-07-13:
Board part Number :  Generic F411RETx

Original program is quite big so I'm sharing the testing program for the EEPROM part.

`const int input = 2;

int CAL_ADDR = 0x10;
int SETUP_ADDR = 0x70;
uint32_t time9 = 0;
long ZEROSCALE = 15000;
long AUTO_ZERO = 5;

void setup()
{
  Serial.begin(9600);     // PC Comm.
  pinMode(2, INPUT_PULLUP);
  EEPROM_read();
}

void loop()
{
  if (digitalRead(2) == LOW)
    EEPROM_write();
}

void EEPROM_write()
{
  EEPROMWritelong(CAL_ADDR, ZEROSCALE);
  EEPROMWritelong(SETUP_ADDR, AUTO_ZERO);
  eeprom_buffer_flush();
}

void EEPROM_read()
{
  eeprom_buffer_fill();
  ZEROSCALE    = EEPROMReadlong(CAL_ADDR);
  AUTO_ZERO    = EEPROMReadlong(SETUP_ADDR);
  Serial.println(ZEROSCALE);
  Serial.println(AUTO_ZERO);
}

void EEPROMWritelong(int address, long value)
{
  byte four  = (value & 0xFF);
  byte three = ((value >> 8) & 0xFF);
  byte two   = ((value >> 16) & 0xFF);
  byte one   = ((value >> 24) & 0xFF);

  eeprom_buffered_write_byte(address, four); delay(1);
  eeprom_buffered_write_byte(address + 1, three); delay(1);
  eeprom_buffered_write_byte(address + 2, two); delay(1);
  eeprom_buffered_write_byte(address + 3, one); delay(1);
}

long EEPROMReadlong(long address)
{
  byte four  = eeprom_buffered_read_byte(address);
  byte three = eeprom_buffered_read_byte(address + 1);
  byte two   = eeprom_buffered_read_byte(address + 2);
  byte one   = eeprom_buffered_read_byte(address + 3);

  return ((four << 0) & 0xFF) + ((three << 8) & 0xFFFF) + ((two << 16) & 0xFFFFFF) + ((one << 24) & 0xFFFFFFFF);
}
`

I tried selecting **Nucleo F411RE** as well but its not compiling. But the same program compiles and works fine with 1.9.0 core.

fpistm, 2021-07-13:
Thanks. I'm surprised you got error as the sketch used by the CI build includes the EEPROM API to avoid such issue.
I've tested to build default EEPROM sketch and had no build issue.
Tested your sketch and it failed because you didn't include EEPROM header:
`#include <EEPROM.h>`

It is a built-in library but you always have to include it like for SPI or Wire.
So simply include the library and this is OK.

closed 2021-07-13 by fpistm

Vicky-S, 2021-07-13:
Is Buffered EEPROM implementation same with Inbuilt EEPROM? Then how its working with 1.9.0 core. I dont know.

fpistm, 2021-07-13:
> 
> 
> Is Buffered EEPROM implementation same with Inbuilt EEPROM? Then how its working with 1.9.0 core. I dont know.

Got it, I moved the bsp inside the EEPROM library:
https://github.com/stm32duino/Arduino_Core_STM32/pull/1152

This is why now it does not built as `stm32_eeprom.h` is no included by default but this is what is expected. If you need EEPROM API you have to include the library. That's why the major core version has changed. 😉

Vicky-S, 2021-07-13:
Ok. Thank You.

HappySamuel, 2021-07-21:
Hi @fpistm 

I have did the eeprom include using `#include <EEPROM.h>` , but my stm32f103c8 / stm32f103cb will still hang if i trigger the EEPROM.write(). However, EEPROM.read() has no issue at all.

What shall i do to get it working?

Best,
Samuel

fpistm, 2021-07-21:
Hi @HappySamuel 
I don't know, I've tested and it's work with a Bluepill.
Flash has a limited number of write cycle so maybe you reach it .