stm32duino/Arduino_Core_STM32 #1447: HID upload for F0 variants

geraldjust, 2021-07-15:
Hi, I saw from Here: https://github.com/stm32duino/Arduino_Core_STM32/issues/1000
you were able to crack a way to flash the firmware using the stock bootloader with a F070. I'm trying to use this: https://github.com/aroerina/STM32_HID_Bootloader_F0
which does the same thing as copying the Vector into ram since it does not have a VTOR register. it seems that i only had to modify the script.ld so make flash and sram to work per the example. Then copy over in main program before HAL_INIT(). I've been trying to get this to work with the stm32duino but have been running into a brick wall. Is there any way you can help me?  i got some pointers from the forum on were to place it but it does not seem to work. Weird thing is that it works just fine in cubeIDE.

fpistm, 2021-07-15:
Hi @geraldjust 
Issue is for reporting error not request support. Moreover you already create a topic on this on the forum.
https://www.stm32duino.com/viewtopic.php?f=58&t=1180

closed 2021-07-15 by fpistm

geraldjust, 2021-07-15:
Sorry though here i might be more promising to get someone who could help, Thanks!

fpistm, 2021-07-15:
You can open a GH discussion here:
https://github.com/stm32duino/Arduino_Core_STM32/discussions/categories/q-a