stm32duino/Arduino_Core_STM32 #1451: Generic L151C8TxA - CDC does not compile

toogooda, 2021-07-23:
**Describe the bug**
L151C8TxA does not compile if you chose any of the CDC options
[image]
This is the error message:
```

C:\Users\User\AppData\Local\Arduino15\packages\STMicroelectronics\hardware\stm32\2.0.0\variants\STM32L1xx\L100C6Ux(A)_L151C(6-8-B)(T-U)x(A)_L152C(6-8-B)(T-U)x(A)\PeripheralPins.c:224:58: error: 'GPIO_AF10_USB' undeclared here (not in a function); did you mean 'GPIO_AF0_SWJ'?
  224 |   {PA_11, USB, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_NOPULL, GPIO_AF10_USB)}, // USB_DM
      |                                                          ^~~~~~~~~~~~~
C:\Users\User\AppData\Local\Arduino15\packages\STMicroelectronics\hardware\stm32\2.0.0\cores\arduino/stm32/PinNamesTypes.h:83:29: note: in definition of macro 'STM_PIN_DEFINE'
   83 |                           ((AFNUM & STM_PIN_AFNUM_MASK) << STM_PIN_AFNUM_SHIFT))
      |                             ^~~~~
C:\Users\User\AppData\Local\Arduino15\packages\STMicroelectronics\hardware\stm32\2.0.0\variants\STM32L1xx\L100C6Ux(A)_L151C(6-8-B)(T-U)x(A)_L152C(6-8-B)(T-U)x(A)\PeripheralPins.c:224:16: note: in expansion of macro 'STM_PIN_DATA'
  224 |   {PA_11, USB, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_NOPULL, GPIO_AF10_USB)}, // USB_DM
      |                ^~~~~~~~~~~~
exit status 1
Error compiling for board Generic STM32L1 series.

```

**To Reproduce**
The code is not relevant to this issues as a blank .ino fill not compile but here is is anyway:
```
#define LED_BUILTIN PA1

void setup() {
  pinMode(LED_BUILTIN, OUTPUT);
  Serial.begin(9600);
}

void loop() {
  digitalWrite(LED_BUILTIN, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(1000);                       // wait for a second
  digitalWrite(LED_BUILTIN, LOW);    // turn the LED off by making the voltage LOW
  delay(1000);                       // wait for a second
  Serial.println("Hello World");
}

```

Steps to reproduce the behavior:
1. Select L151c8txA (or probably other similar ones like L151c8tx)
2. Select a CDC option also
3. Hit compile

**Expected behavior**
Compiles

**Desktop (please complete the following information):**
 - OS: Windows
 - Arduino IDE version: [e.g. 1.8.8] 1.8.9 tested on 2.0 also
 - STM32 core version: 2.0 download day previously
 - Tools menu settings if not the default:
 [image]
 - Upload method: SWD

**Board (please complete the following information):**
 - Name: custom
 - Hardware Revision:N/A
 - Extra hardware used if any:N/A
 
RESOLUTION:

I have solved this with a hack but it needs to be fixed properly in the core:
In the following file :
C:\Users\User\AppData\Local\Arduino15\packages\STMicroelectronics\hardware\stm32\2.0.0\variants\STM32L1xx\L100C6Ux(A)_L151C(6-8-B)(T-U)x(A)_L152C(6-8-B)(T-U)x(A)\PeripheralPins.c

I changed:

[image]

I copied this from a working mcu (F103re) but had to replace AFIO_NONE which was undefined with 0 as it was undefined and I eventually found it was element 0 of an enum...

It now shows up as Com22 and works for Serial but I am sure there is a more eloquent permanent fix for the core. 

Happy to help with testing if you make a core change.

fpistm, 2021-07-23:
Hi @toogooda 
Thanks for the report. This is a known issue.
PeripheralPins.c files are automatically generated from the [STM32_open_pin_data](https://github.com/STMicroelectronics/STM32_open_pin_data). I've opened an issue to remove this wrong AF:
https://github.com/STMicroelectronics/STM32_open_pin_data/issues/2

I've fixed for the `L151RET_L152RET_L162RET`:
https://github.com/stm32duino/Arduino_Core_STM32/blob/0686e2d5a873e16842e9f61dcb69fd095893e8f2/variants/STM32L1xx/L151RET_L152RET_L162RET/PeripheralPins.c#L241-L243

But forgot the fix one you pointed.
The fix will be automatically populated when the [STM32_open_pin_data](https://github.com/STMicroelectronics/STM32_open_pin_data) will be updated.

toogooda, 2021-07-23:
@fpistm Perfect :)

I must say I am impressed that all the pins have now been pre generated I am now considering many other chips for prototyping, it doesn't look difficult to get new ones going.

Keep up the good work, STM is becoming more relevant in the prototyping space!

PR #1466 (2021-08-11): variants: L1: update generated files to STM32_open_pin_data v6.0.30
Using a fix to remove `GPIO_AF10_USB` from all L1 GPIO config files. (See https://github.com/STMicroelectronics/STM32_open_pin_data/issues/2)

Fixes #1451
commit 9f34a6e2 (2021-08-11): variants: L1: update generated files to STM32_open_pin_data v6.0.30

Using a fix to remove GPIO_AF10_USB from all L1 GPIO config files.

Fixes #1451
closed 2021-08-11 by fpistm