stm32duino/Arduino_Core_STM32 #1458: Missing errata fix in stm32l4xx_hal_sd.c

Bambofy, 2021-08-04:
**Describe the bug**
The bug is that some v1.0 SD cards will not initialize properly.

The bug is described the errata sheet at section 2.15.2 "Wrong CCRCFAIL status after a response without CRC is received" here https://www.st.com/resource/en/errata_sheet/es0388-stm32l452xx-device-errata-stmicroelectronics.pdf

The bug occurs at this line of code https://github.com/stm32duino/Arduino_Core_STM32/blob/master/system/Drivers/STM32L4xx_HAL_Driver/Src/stm32l4xx_hal_sd.c#L3546 according to the errata i believe this IF statement needs modifying.

**To Reproduce**
Connect a version 1.0 Sd card into an SDMMC slot on an STM32L452RE and attempt to use STM32SD to initialize it.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. Windows]
 - Arduino IDE version: [e.g. 1.8.8]
 - STM32 core version: [e.g. 1.4.0]
 - Tools menu settings if not the default: [e.g. Newlib Standard, No Serial]
 - Upload method: [e.g. SWD]

**Board (please complete the following information):**
 - Name: Nucleo-64

Bambofy, 2021-08-04:
Found that it was just a broken SD card.

fpistm, 2021-08-09:
Hi  @Bambofy 

The core uses only the HAL and avoid changes.
You should open an issue to the dedicated repository then you will have a feedback from ST HAL Team about this issue: https://github.com/STMicroelectronics/STM32CubeL4/issues
Then if a fix is made in a further release it will be integrated when the HAL driver will be updated in the core.

closed 2021-08-09 by fpistm