stm32duino/Arduino_Core_STM32 #1471: Support for STM32L152CBT6[New Variant]

forfrends, 2021-08-12:
- Name: STM32L152CBT6
 - Link to all relevant information:
Documentations: [https://www.st.com/en/microcontrollers-microprocessors/stm32l152cb.html](https://www.st.com/en/microcontrollers-microprocessors/stm32l152cb.html)
[https://www.st.com/resource/en/datasheet/stm32l152cb.pdf](https://www.st.com/resource/en/datasheet/stm32l152cb.pdf)

Hello. Can you add support for STM32L152CBT6? This is a very interesting chip. It is used in many applications where low power consumption is required.

fpistm, 2021-08-12:
Hi @forfrends 
Since STM32 core version 2.0.0 the STM32L152CBT is available as a generic target:
https://github.com/stm32duino/Arduino_Core_STM32/blob/f2cec8cf3045c1df58d1effb6fd7ac63110b129c/boards.txt#L4378-L4384

closed 2021-08-12 by fpistm

forfrends, 2021-08-12:
O! Thank you! Something I missed it