stm32duino/Arduino_Core_STM32 #1473: STM32WL platformio options for cortex-m4 turn on fpu, but STM32WL55 CMSIS drivers define __FPU_PRESENT=0

dolfandringa, 2021-08-15:
**Describe the bug**
See also https://www.stm32duino.com/viewtopic.php?p=8039#p8039

When compiling code for the Nucleo STM32WL55JC1 with the Arduino Core STM32 v2.0.0, platformio automatically adds the `-mfpu=fpv4-sp-d16", "-mfloat-abi=hard` options because the mcu is defined as cortex-m4. This results in an error because  the CMSIS driver defines `__FPU_PRESENT=0`.

**To Reproduce**

1. Use this branch (WIP) for platformio which adds support for the STM32WL55 board using the arduino framework: https://github.com/dolfandringa/platform-ststm32/tree/feature/stm32duino_stm32wl55_support
2. Make a new project using the platformio.ini below. No need for any code, just use an empty main.cpp like below.
3. run `pio run` to compile the code. 

`platformio.ini`
```
[env:stm32wl55]
platform = ststm32
board = nucleo_wl55jc
framework = arduino
```
`src/main.cpp`
```
#include <Arduino.h>

void setup() {
}

void loop() {
}
```

this results in the following error:
```
In file included from /home/dolfandringa/.platformio/packages/framework-arduinoststm32/system/Drivers/CMSIS/Device/ST/STM32WLxx/Include/stm32wl55xx.h:210,
                 from /home/dolfandringa/.platformio/packages/framework-arduinoststm32/system/Drivers/CMSIS/Device/ST/STM32WLxx/Include/stm32wlxx.h:99,
                 from /home/dolfandringa/.platformio/packages/framework-arduinoststm32/cores/arduino/stm32/stm32_def.h:56,
                 from /home/dolfandringa/.platformio/packages/framework-arduinoststm32/cores/arduino/stm32/clock.h:19,
                 from /home/dolfandringa/.platformio/packages/framework-arduinoststm32/cores/arduino/wiring_time.h:23,
                 from /home/dolfandringa/.platformio/packages/framework-arduinoststm32/cores/arduino/wiring.h:38,
                 from /home/dolfandringa/.platformio/packages/framework-arduinoststm32/cores/arduino/Arduino.h:36,
                 from /home/dolfandringa/.platformio/packages/framework-arduinoststm32/variants/STM32WLxx/WL54JCI_WL55JCI_WLE4J(8-B-C)I_WLE5J(8-B-C)I/PeripheralPins.c:21:
/home/dolfandringa/.platformio/packages/framework-cmsis/CMSIS/Core/Include/core_cm4.h:105:8: error: #error "Compiler generates FPU instructions for a device without an FPU (check __FPU_PRESENT)"
  105 |       #error "Compiler generates FPU instructions for a device without an FPU (check __FPU_PRESENT)"
```

**Expected behavior**
The code should compile correctly.

**Desktop (please complete the following information):**
 - OS: Fedora 34 linux
 - Arduino IDE version: latest platformio master branch with fork in https://github.com/dolfandringa/platform-ststm32/tree/feature/stm32duino_stm32wl55_support
 - STM32 core version: master branch from git/2.0.0

**Board (please complete the following information):**
 - Name: Nucleo WL55JC1

dolfandringa, 2021-08-15:
I am working on a resolution for this issue, and any other platformio related fixes in https://github.com/dolfandringa/Arduino_Core_STM32/tree/feature/support_stm32wl55jc1_on_platformio

fpistm, 2021-08-15:
Hi @dolfandringa 
We do not support PIO here.
Moreover there is no FPU for the STM32WL. For coretx M4, Floating-Point Unit (FPU) is optional, see:
https://www.st.com/content/st_com/en/arm-32-bit-microcontrollers/arm-cortex-m4.html

CMSIS define is correct.

dolfandringa, 2021-08-15:
There is a bug in your repository in https://github.com/stm32duino/Arduino_Core_STM32/blob/master/tools/platformio/platformio-build.py. That is what my code addresses, a file in your repository. I am also contributing a PR to PIO.

fpistm, 2021-08-15:
> There is a bug in your repository in https://github.com/stm32duino/Arduino_Core_STM32/blob/master/tools/platformio/platformio-build.py. That is a file you are clearlyt maintaining, since it is in this repository, and I am very happy to help with a fix.

In fact no, it was provided by @valeros from pio to allow CI build that's all. 😉

dolfandringa, 2021-08-15:
yes, but it is in your repository, so can only be fixed there, right? Its a very simple change that I just submitted a PR for.

fpistm, 2021-08-15:
Honestly, I thought it was only used for the CI PIO build not by the framework itself.
Hi @valeros please could you comment on this? Thanks in advance.

closed 2021-08-16 by fpistm

ztliu0722, 2022-07-25:
> Hi @dolfandringa We do not support PIO here. Moreover there is no FPU for the STM32WL. For coretx M4, Floating-Point Unit (FPU) is optional, see: https://www.st.com/content/st_com/en/arm-32-bit-microcontrollers/arm-cortex-m4.html
> 
> CMSIS define is correct.

Thanks for clarifying that there is NO FPU in STM32WL core here. 
I misconceived that FPU is implemented in every  STM32 CM4 microcontroller.
For in[ PM0214 STM32 Cortex®-M4 MCUs and MPUs programming manual](https://www.st.com/resource/en/programming_manual/pm0214-stm32-cortexm4-mcus-and-mpus-programming-manual-stmicroelectronics.pdf), section 1.3.3, it's written:
"_IEEE754-compliant single-precision FPU implemented in all STM32 Cortex-M4 microcontrollers_".
That made me conceive that every ST's CM4 core has FPU implemented.

PR #1474 (2021-08-16): exclude fpu options from stm32wl55
**Summary**

This PR fixes **bug**

* [x] #1473 

Arduino Core STM32 provides a platformio-build.py file that sets compiler build flags for Arduino Core STM32 in platformio. This file sets the -mfpu flag automatically for every cortex-m4 MCU. But according to the CMSIS drivers, the STM32WL55 does not contain an FPU, causing this python script to set the wrong compile time flags. The proposed PR fixes this by excluding the -mfpu option from the CPPDEFINES flags. 

**Validation**

* Ensure CI build is passed.
The following sketch won't compile without it, but w [... cut]