stm32duino/Arduino_Core_STM32 #1477: I2C issue between STM32F411CE and Jetson nano

camelator, 2021-08-19:
I am using platformio (last version).
here are the package lists embeded to my simple code:
- framework-arduinoststm32 4.20000.210603 (2.0.0) 
 - framework-cmsis 2.50700.210515 (5.7.0) 
 - tool-dfuutil 1.9.200310 
 - tool-openocd 2.1100.0 (11.0) 
 - tool-stm32duino 1.0.1 

I just tried to make sure the I2C connection between my STM32F411CE and jetson nano is working. 
The command I am using on jetson nano is sudo i2cdetect -r -y 1
on the same I2C line I can add a servo controller and it is perfectly detected by Jetson Nano => I assume Jetson Nano is working fine.

I have 4.7K pull-up resistors on both lines.
But unfortunately, the STM32 is never detected.
I am using PA6 and PA7 as SDA and SCL.

Any help????

#include <Arduino.h>
#include <Wire.h>
#define SLAVE_SDA PA6
#define SLAVE_SCL PA7
#define SLAVE_ADR 0x4

TwoWire slaveI2C (SLAVE_SDA,SLAVE_SCL) ;

void receiveEvent (int) ;

void setup() {
  // put your setup code here, to run once:

  Serial.begin(115200) ;
  slaveI2C.begin(SLAVE_ADR) ;
  slaveI2C.onReceive(receiveEvent) ;
    
}

void loop() {

    delay(100) ;
  
}

void receiveEvent (int howMany) {

  if (slaveI2C.available()<1) 
    return ;

  uint8_t command=slaveI2C.read() ;

}

fpistm, 2021-08-19:
Hi @camelator 
Unfortunately, we do not support issue met with PIO only with Arduino IDE.
Moreover, such kind of support should be requested on the forum.

Finally, STM32F411C has no I2C capabilities on pins PA6/PA7:
https://github.com/stm32duino/Arduino_Core_STM32/blob/fa8bebdd052ba445c6f442d415744619633cd745/variants/STM32F4xx/F411C(C-E)(U-Y)/PeripheralPins.c#L52-L74

You can also check the datasheet:
https://www.st.com/resource/en/datasheet/stm32f411ce.pdf

[image]

closed 2021-08-19 by fpistm

camelator, 2021-08-19:
Just tried with PB7 as SDA and PB6 as SCLK.
I am using ArduinoIDE and please see below my sketch.
Errors produced by i2cdetect on Jetson Nano:
command: I2Cdetect -r -y 1
the command becomes sow. After CTRL+C then dmesg:
no acknowledge from address 0x3
[ 7843.509985] tegra-i2c 7000c400.i2c: pio timed out addr: 0x4 tlen:12 rlen:4
[ 7843.517357] tegra-i2c 7000c400.i2c: --- register dump for debugging ----
[ 7843.527385] tegra-i2c 7000c400.i2c: I2C_CNFG - 0x22c00
[ 7843.533331] tegra-i2c 7000c400.i2c: I2C_PACKET_TRANSFER_STATUS - 0x10001
[ 7843.540243] tegra-i2c 7000c400.i2c: I2C_FIFO_CONTROL - 0xe0
[ 7843.546020] tegra-i2c 7000c400.i2c: I2C_FIFO_STATUS - 0x800080
[ 7843.552002] tegra-i2c 7000c400.i2c: I2C_INT_MASK - 0x7d
[ 7843.557351] tegra-i2c 7000c400.i2c: I2C_INT_STATUS - 0x2
[ 7843.562806] tegra-i2c 7000c400.i2c: i2c transfer timed out addr: 0x4
[ 7853.749749] tegra-i2c 7000c400.i2c: pio timed out addr: 0x5 tlen:12 rlen:4

Please can you help?

#include <Arduino.h>
#include <Wire.h>

#define SLAVE_SDA PB7
#define SLAVE_SCL PB6
#define SLAVE_ADR 0x4

TwoWire slaveI2C (SLAVE_SDA,SLAVE_SCL) ;

void receiveEvent (int) ;

void setup() {
  // put your setup code here, to run once:
  /*
  pinMode ( LEFT_MOTOR_NEG_CTRL_PIN, OUTPUT );
  pinMode ( LEFT_MOTOR_NEG_PWM_PIN, OUTPUT );
  pinMode ( LEFT_MOTOR_POS_CTRL_PIN, OUTPUT );
  pinMode ( LEFT_MOTOR_POS_PWM_PIN, OUTPUT );
  digitalWrite(LEFT_MOTOR_NEG_CTRL_PIN,HIGH) ;
  digitalWrite(LEFT_MOTOR_POS_CTRL_PIN,HIGH) ;

  pinMode ( RIGHT_MOTOR_NEG_CTRL_PIN, OUTPUT );
  pinMode ( RIGHT_MOTOR_NEG_PWM_PIN, OUTPUT );
  pinMode ( RIGHT_MOTOR_POS_CTRL_PIN, OUTPUT );
  pinMode ( RIGHT_MOTOR_POS_PWM_PIN, OUTPUT );
  digitalWrite(RIGHT_MOTOR_NEG_CTRL_PIN,HIGH) ;
  digitalWrite(RIGHT_MOTOR_POS_CTRL_PIN,HIGH) ;
*/

  Serial.begin(115200) ;
  //leftDriver.init() ;
  //rightDriver.init() ;
  slaveI2C.setClock(400000) ;
  slaveI2C.begin(SLAVE_ADR) ;
  slaveI2C.onReceive(receiveEvent) ;
    
  //Wire.begin(SLAVE_ADR) ;

}

void loop() {
  // put your main code here, to run repeatedly:
  /*analogWrite(LEFT_MOTOR_NEG_PWM_PIN,0) ;
  analogWrite(LEFT_MOTOR_POS_PWM_PIN,32) ;
  analogWrite(RIGHT_MOTOR_NEG_PWM_PIN,0) ;
  analogWrite(RIGHT_MOTOR_POS_PWM_PIN,32) ;
  delay(500) ;
  analogWrite(LEFT_MOTOR_NEG_PWM_PIN,0) ;
  analogWrite(LEFT_MOTOR_POS_PWM_PIN,0) ;
  analogWrite(RIGHT_MOTOR_NEG_PWM_PIN,0) ;
  analogWrite(RIGHT_MOTOR_POS_PWM_PIN,0) ;
  while (true) {
    delay(1000) ;
  }

  analogWrite(RIGHT_MOTOR_NEG_PWM_PIN,0) ;
  analogWrite(RIGHT_MOTOR_POS_PWM_PIN,0) ;
*/

  while (true) {
    delay(100) ;

  }
  
}

void receiveEvent (int howMany) {

  if (slaveI2C.available()<1) 
    return ;

  uint8_t command=slaveI2C.read() ;
 

}

LMESTM, 2021-08-20:
For generic usage questions and support request please use the forum : https://www.stm32duino.com/
The github issues are for core issues that have been cornered a minimum :-) 
thanks for your understanding