stm32duino/Arduino_Core_STM32 #1479: I2C issue between STM32F411CE and Jetson nano

camelator, 2021-08-19:
I am using ArduinoIDE  1.8.15
The I2C pins used are PB7 (SDA) and PB6 (SCL)
I am trying an I2C connection between my STM32F411CE (slave) and jetson nano (master) as I completed here (but I can not change the status of the issue):
https://github.com/stm32duino/Arduino_Core_STM32/issues/1477

The command I am using on jetson nano is sudo i2cdetect -r -y 1
on the same I2C line I can add a servo controller and it is perfectly detected by Jetson Nano => I assume Jetson Nano is working fine.
I have 4.7K pull-up resistors on both lines.

Having STM32 plugged a strange behavior happears : the I2cdetect command becomes slow at index 4, and I have to interrupt the command by CTRL+C
Here are the error message found with dmesg command:
no acknowledge from address 0x3
[ 7843.509985] tegra-i2c 7000c400.i2c: pio timed out addr: 0x4 tlen:12 rlen:4
[ 7843.517357] tegra-i2c 7000c400.i2c: --- register dump for debugging ----
[ 7843.527385] tegra-i2c 7000c400.i2c: I2C_CNFG - 0x22c00
[ 7843.533331] tegra-i2c 7000c400.i2c: I2C_PACKET_TRANSFER_STATUS - 0x10001
[ 7843.540243] tegra-i2c 7000c400.i2c: I2C_FIFO_CONTROL - 0xe0
[ 7843.546020] tegra-i2c 7000c400.i2c: I2C_FIFO_STATUS - 0x800080
[ 7843.552002] tegra-i2c 7000c400.i2c: I2C_INT_MASK - 0x7d
[ 7843.557351] tegra-i2c 7000c400.i2c: I2C_INT_STATUS - 0x2
[ 7843.562806] tegra-i2c 7000c400.i2c: i2c transfer timed out addr: 0x4
[ 7853.749749] tegra-i2c 7000c400.i2c: pio timed out addr: 0x5 tlen:12 rlen:4

Here is my ino file:

#include <Arduino.h>
#include <Wire.h>

#define SLAVE_SDA PB7
#define SLAVE_SCL PB6
#define SLAVE_ADR 0x4

TwoWire slaveI2C (SLAVE_SDA,SLAVE_SCL) ;

void receiveEvent (int) ;

void setup() {
  // put your setup code here, to run once:
  /*
  pinMode ( LEFT_MOTOR_NEG_CTRL_PIN, OUTPUT );
  pinMode ( LEFT_MOTOR_NEG_PWM_PIN, OUTPUT );
  pinMode ( LEFT_MOTOR_POS_CTRL_PIN, OUTPUT );
  pinMode ( LEFT_MOTOR_POS_PWM_PIN, OUTPUT );
  digitalWrite(LEFT_MOTOR_NEG_CTRL_PIN,HIGH) ;
  digitalWrite(LEFT_MOTOR_POS_CTRL_PIN,HIGH) ;

  pinMode ( RIGHT_MOTOR_NEG_CTRL_PIN, OUTPUT );
  pinMode ( RIGHT_MOTOR_NEG_PWM_PIN, OUTPUT );
  pinMode ( RIGHT_MOTOR_POS_CTRL_PIN, OUTPUT );
  pinMode ( RIGHT_MOTOR_POS_PWM_PIN, OUTPUT );
  digitalWrite(RIGHT_MOTOR_NEG_CTRL_PIN,HIGH) ;
  digitalWrite(RIGHT_MOTOR_POS_CTRL_PIN,HIGH) ;
*/

  Serial.begin(115200) ;
  //leftDriver.init() ;
  //rightDriver.init() ;
  slaveI2C.setClock(400000) ;
  slaveI2C.begin(SLAVE_ADR) ;
  slaveI2C.onReceive(receiveEvent) ;
    
  //Wire.begin(SLAVE_ADR) ;

}

void loop() {
  // put your main code here, to run repeatedly:
  /*analogWrite(LEFT_MOTOR_NEG_PWM_PIN,0) ;
  analogWrite(LEFT_MOTOR_POS_PWM_PIN,32) ;
  analogWrite(RIGHT_MOTOR_NEG_PWM_PIN,0) ;
  analogWrite(RIGHT_MOTOR_POS_PWM_PIN,32) ;
  delay(500) ;
  analogWrite(LEFT_MOTOR_NEG_PWM_PIN,0) ;
  analogWrite(LEFT_MOTOR_POS_PWM_PIN,0) ;
  analogWrite(RIGHT_MOTOR_NEG_PWM_PIN,0) ;
  analogWrite(RIGHT_MOTOR_POS_PWM_PIN,0) ;
  while (true) {
    delay(1000) ;
  }

  analogWrite(RIGHT_MOTOR_NEG_PWM_PIN,0) ;
  analogWrite(RIGHT_MOTOR_POS_PWM_PIN,0) ;
*/

  while (true) {
    delay(100) ;

  }
  
}

void receiveEvent (int howMany) {

  if (slaveI2C.available()<1) 
    return ;

  uint8_t command=slaveI2C.read() ;
 

}

fpistm, 2021-08-20:
As stated by me in #1477 and by @LMESTM:
https://github.com/stm32duino/Arduino_Core_STM32/issues/1477#issuecomment-902482887

So please use the forum maybe someone already experiment with Jetson nano.

closed 2021-08-20 by fpistm

fpistm, 2021-08-20:
For ref: https://www.stm32duino.com/viewtopic.php?f=24&t=1223