stm32duino/Arduino_Core_STM32 #1483: missing HardFault_Handler()

Fleckz, 2021-08-23:
When HardFault occures - we end up in WWDG_IRQHandler(), would be great to have custom HardFault_Handler functions to save state for debugging. I am using H743VIT6 in this case, but happens also with F411CEU6

**To Reproduce**
Call this function in your sketch and debug...:

uint32_t read_from_bad_address(void) {
  return *(volatile uint32_t *)0xbadcafe;
}

**Result**
(gdb) bt
#0  0x08004dc4 in WWDG_IRQHandler ()
#1  <signal handler called>
#2  0x08000372 in read_from_bad_address () at /home/user/Arduino/hf_test/hf_test.ino:14
#3  0x080003b4 in loop () at /home/user/Arduino/hf_test/hf_test.ino:31

fpistm, 2021-08-24:
Hi @Fleckz
You can override the Default Hard Fault_Handler, as all the exceptions handlers are build with “Weak” linkage in CMSIS, it is very easy to create your own Hard Fault handler. Simply define a function with the name “HardFault_Handler”, as in:

```
void HardFault_Handler(void)
{ 
   while(1); 
}
```

If you define it in the ino file as it is converted in cpp the you have to use `extern "C"` before.
Hereafter a small example using Nucleo H743ZI2 with `LED_BUILTIN `on `PB0`

```C++
extern "C" void HardFault_Handler(void)
{
#ifdef LED_BUILTIN
  GPIO_TypeDef *gpio = (GPIO_TypeDef *)GPIOB_BASE;
  __disable_irq();
  __HAL_RCC_GPIOB_CLK_ENABLE();
  LL_GPIO_SetPinMode(gpio, LL_GPIO_PIN_0, LL_GPIO_MODE_OUTPUT);
  for (;;) {
    int i;
    for (i = 0; i < 8; i++) {
      LL_GPIO_TogglePin(gpio, LL_GPIO_PIN_0);
      delayMicroseconds(300000);
    }
    delayMicroseconds(200000000);
  }
#else
  while (1);
#endif // LED_BUILTIN
}

void setup() {
  // put your setup code here, to run once:
  read_from_bad_address();
}

void loop() {
  // put your main code here, to run repeatedly:

}

uint32_t read_from_bad_address(void) {
  return *(volatile uint32_t *)0xbadcafe;
}
```

closed 2021-08-24 by fpistm

Fleckz, 2021-08-24:
Ohh, so my problem was, I din't have extern "C", hmm, I thought I tried that also, but I guess I did not! Thank you! My bad!