stm32duino/Arduino_Core_STM32 #1484: Compiler warnings in Arduino IDE for v2.0.0

zfields, 2021-08-23:
**Describe the bug**
Compiler warnings are generated when "Compiler warnings: All" is enabled.

```
In file included from /home/zak/.arduino15/packages/STMicroelectronics/hardware/stm32/2.0.0/libraries/SrcWrapper/src/LL/stm32yyxx_ll_adc.c:35:
/home/zak/.arduino15/packages/STMicroelectronics/hardware/stm32/2.0.0/system/Drivers/STM32L4xx_HAL_Driver/Src/stm32l4xx_ll_adc.c: In function 'LL_ADC_CommonDeInit':
/home/zak/.arduino15/packages/STMicroelectronics/hardware/stm32/2.0.0/system/Drivers/STM32L4xx_HAL_Driver/Src/stm32l4xx_ll_adc.c:310:53: warning: unused parameter 'ADCxy_COMMON' [-Wunused-parameter]
  310 | ErrorStatus LL_ADC_CommonDeInit(ADC_Common_TypeDef *ADCxy_COMMON)
      |                                 ~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~

In file included from /home/zak/.arduino15/packages/STMicroelectronics/hardware/stm32/2.0.0/libraries/SrcWrapper/src/LL/stm32yyxx_ll_rng.c:23:
/home/zak/.arduino15/packages/STMicroelectronics/hardware/stm32/2.0.0/system/Drivers/STM32L4xx_HAL_Driver/Src/stm32l4xx_ll_rng.c: In function 'LL_RNG_DeInit':
/home/zak/.arduino15/packages/STMicroelectronics/hardware/stm32/2.0.0/system/Drivers/STM32L4xx_HAL_Driver/Src/stm32l4xx_ll_rng.c:87:40: warning: unused parameter 'RNGx' [-Wunused-parameter]
   87 | ErrorStatus LL_RNG_DeInit(RNG_TypeDef *RNGx)
      |                           ~~~~~~~~~~~~~^~~~
```

**To Reproduce**
Observed when using the standard `Servo.h` library.

Steps to reproduce the behavior:
1. Select an example sketch from the Servo library
2. Select "Compiler warnings: All" from the Arduino IDE preferences window.
3. Attempt to compile the selected example
4. See warnings

**Expected behavior**
No warnings should be generated from the board support library.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Pop_OS! 20.04
 - Arduino IDE version: [e.g. 1.8.13]
 - STM32 core version: [e.g. 2.0.0]
 - Upload method: N/A

**Board (please complete the following information):**
 - Name: Generic STM32L4 series
 - Hardware Revision: 
 - Extra hardware used if any: 

**Additional context**

fpistm, 2021-08-24:
Hi @zfields 
I know HAL/LL raised several warnings. Anyway, the STM32 core simply uses them "as it" and do not made any changes.
I've made a small update to the wrapper to ignore them:
https://github.com/stm32duino/Arduino_Core_STM32/commit/aa8ec6bcf80a35fb398a576003882dd40c0de780
https://github.com/stm32duino/Arduino_Core_STM32/commit/23dab9eb685d139817bae68067b707237c8f51bc

 but in the case you point the LL header is included but the HAL itself that's why warnings are raised.

Those warnings should be fixed in the STM32 Cube L4, you could open an issue here: https://github.com/STMicroelectronics/STM32CubeL4/issues

Then when it will be fixed and release this will benefit to all users of this cube FW including the STM32 core when it will be updated.

closed 2021-08-24 by fpistm

zfields, 2021-08-24:
This issue has been reopened on `STMicroelectronics/STM32CubeL4` as requested: https://github.com/STMicroelectronics/STM32CubeL4/issues/44