stm32duino/Arduino_Core_STM32 #1495: Double deploy required to flash Swan_R5

zfields, 2021-09-15:
**Describe the bug**
When deploying a sketch to the Swan (STM32L4xx), the first deploy seems to hang in the bootloader and the sketch will not load. If a redeploy is attempted, then the sketch will load as expected. This reoccurs every time a change is made to the sketch code.

**To Reproduce**
Use the `Blink` example from the Arduino Examples folder.

Steps to reproduce the behavior:
1. Using the Swan (or perhaps any Generic STM32L4xx)
2. Flash the Blink app using the Arduino IDE or VSCode
3. See error (no blinking)

**Expected behavior**
I would expect the device to start the sketch once it has been programmed.

**Desktop (please complete the following information):**
 - OS: Verified on Mac, Linux and Windows
 - Arduino IDE version: 1.8.13
 - STM32 core version: 2.0.0 (@ master)
 - Tools menu settings if not the default: C Runtime Library - Newlib Nano (default), Debug Symbols - Enabled (-g), Optimize - Debug (-Og), USB Speed - Low/Full Speed, USB Support - CDC (generic `Serial` supercede U(S)ART), U(S)ART support: - Enabled (generic `Serial`), 
 - Upload method: SWD

**Board (please complete the following information):**
 - Name: Generic STM32L4 series
 - Hardware Revision: Swan R5

**Additional context**
Modifying line `5070` of `~/.arduino15/packages/STMicroelectronics/hardware/stm32/2.0.0/boards.txt`, to read:
```none
GenL4.menu.upload_method.swdMethod.upload.protocol=10
```
opposed to the original:
```none
GenL4.menu.upload_method.swdMethod.upload.protocol=0
```
... seems to have fixed the problem.

Can you explain what is happening and explain if there is a better and/or more targeted option for the Swan?

fpistm, 2021-09-16:
Hi @zfields 
This issue has been reported by @bsatrom by mail.
Opening an issue will not help as now 2 channels are now opened on the same subject.
Just a thought, did you update your STLink FW?

zfields, 2021-09-16:
No, I have not updated the firmware on my STLink.

Let's close this issue for now, and maybe update the thread later if issue appears to be more pervasive than the Swan_R5.

closed 2021-09-16 by zfields

zfields, 2021-09-16:
To close the loop.

I just performed a firmware upgrade on my STLink-V3 to bring it to the latest firmware, and the problem still persists.

However, the change to the `boards.txt` file, still provides the desired outcome.

fpistm, 2021-09-16:
Yes, anyway the change is not valid as it impacts all the GenL4 targets.

PR #1511 (2021-09-30): Move Swan board to dedicated menu and add a workaround for SWD upload method
This PR is a workaround to allow flashing the Swan board at once. An issue is under investigations by ST BL team.
While waiting to have their return adding `-ob nBOOT0=0 -ob nBOOT0=1` to the upload options allows to properly flash and run the application.

Fixes #1495

During the test I found an issue with the STM32cubeProgrammer script due to the use of '=' in option.
I've push a fix in the [stm32duino/Arduino_Tools](https://github.com/stm32duino/Arduino_Tools) repo: https://github.com/stm32duino/Arduino_Tools/commit [... cut]
commit 64437045 (2021-09-30): fix(swan): workaround to flash and start application at once

when using the SWD interface.

Fixes #1495