stm32duino/Arduino_Core_STM32 #1516: Nucleo-L4R5 is kept in reset after downloading sketch

Pestimist, 2021-10-15:
Hi,

I am experimenting with Nucleo-L4R5 and try to run the [CheckVariant example sketch](https://github.com/stm32duino/STM32Examples/tree/main/examples/NonReg/CheckVariant) on it.
This sometime succeedes, but most of the time the µC gets stuck in Reset. The red LED is on. When trying to program it with STM32CubeProgramer I get the error "The core is kept under Reset!"
Doing a full chip erase and several resets is able to bring the µC back to normal operation but this just keeps happening very often.
I also tried downloading a simple "blink" example, same result.
When connecting the stuck board to STM32CubeProgrammer, fault analysis is executed and I get the message that DIV_0_TRP bit was enabled.
I suspect this is an error with stm32duino, not with the µC itself, thats why I am writing here.

fpistm, 2021-10-15:
Unfortunately not it is too specific to be added to generic upload method.

Pestimist, 2021-10-15:
too bad
can I make use of the fix you proposed? How could I do that?

fpistm, 2021-10-15:
Well you can locally change the GenL4 upload line to add: -ob nBOOT0=0 -ob nBOOT0=1
https://github.com/stm32duino/Arduino_Core_STM32/blob/6479b8a0936852c052c16f79606cd1192fc9dcc1/boards.txt#L5089
I did not add it as issue seems linked to STM32L4+ and do not want impact all the STM32L4.

Pestimist, 2021-10-15:
After successfully downloading the CheckVariant sketch via SWD the µC started to behave very erratically.

First it was unreacheable, found out that Flash protection registers were changed.
Managed to get them back to normal and program the chip anew.
But the registers keep getting messed up on nearly every programming.
And the chip keeps resetting, output on terminal is as follows:
```
#####
NUM_DIGITAL_PINS#####
NUM_DIGITAL_PINS#####
NUM_DIGITAL_PINS#####
NUM_DIGITAL_PINS#####
NUM_DIGITAL_PINS#####
NUM_DIGITAL_PINS#####
NUM_DIGITAL_PINS#####
NUM_DIGITAL_PINS#####
NUM_DIGITAL_PINS#####
NUM_DIGITAL_PINS#####
NUM_DIGITAL_PINS#####
```

This happens regardless how I program the chip (Arduino IDE, Sloeber, STM32programmer, mass storage, SWD)
Even writing registers and resetting the core within STM32Cube Programmer yields the wrong entries in the registers.
seems like the chip itself is broken?

Pestimist, 2021-10-15:
I wanted to add the L4P5 variant, did all the steps mentioned [here](https://github.com/stm32duino/wiki/wiki/Add-a-new-variant-%28board%29) but had no serial output after programming the CheckVariant sketch.
So I tried the sketch wit the L4Rvariant just to make sure all the rest of my setup is working fine.
That is how I ran into all these issues.

Pestimist, 2021-10-18:
ST-Link firmware update did not change anything.
To be honest I don't really know much about the option bytes, and there are many :)
Anything special you consider worth looking at?
Thanks for your support!

Pestimist, 2021-10-18:
[image]
[image]
[image]

Pestimist, 2021-10-18:
side question:
is there a difference between Windows and Linux versions of Arduino IDE and STM32duino considering this issue?
I'm asking because build on Linux is way faster even when running out of a VM...
would save me some time