stm32duino/Arduino_Core_STM32 #1579: Installating v2.1.0 over older version in Ubuntu fails

jpmeijers, 2021-12-05:
Arduino board manager listed stm32duino as deprecated and  required changing the boards URL. After changing the URL and trying to install the latest version on stm32duino results in an error:

```
 /home/jpmeijers/.arduino15/packages/STMicroelectronics/tools/xpack-arm-none-eabi-gcc/10.2.1-1.1
java.lang.RuntimeException: java.nio.file.FileAlreadyExistsException: /home/jpmeijers/.arduino15/packages/STMicroelectronics/tools/xpack-arm-none-eabi-gcc/10.2.1-1.1
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$2(ContributionManagerUI.java:175)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.nio.file.FileAlreadyExistsException: /home/jpmeijers/.arduino15/packages/STMicroelectronics/tools/xpack-arm-none-eabi-gcc/10.2.1-1.1
	at sun.nio.fs.UnixException.translateToIOException(UnixException.java:88)
	at sun.nio.fs.UnixException.rethrowAsIOException(UnixException.java:102)
	at sun.nio.fs.UnixException.rethrowAsIOException(UnixException.java:107)
	at sun.nio.fs.UnixFileSystemProvider.createDirectory(UnixFileSystemProvider.java:384)
	at java.nio.file.Files.createDirectory(Files.java:674)
	at java.nio.file.Files.createAndCheckIsDirectory(Files.java:781)
	at java.nio.file.Files.createDirectories(Files.java:727)
	at cc.arduino.contributions.packages.ContributionInstaller.install(ContributionInstaller.java:142)
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$2(ContributionManagerUI.java:172)
	... 1 more
```

I proceeded to manually delete the old board definition folder:
Delete `/home/jpmeijers/.arduino15/packages/STMicroelectronics`

Trying to install again works.

fpistm, 2021-12-05:
Hi @jpmeijers 
this is related to Ardunio IDE which do not refrresh properly the index json. Maybe a restart of the IDE after url change could solve this issue. unfortunately, nothing could be done on our side about this.
Cleaning the package index manually solve your issue, this is fine.

closed 2021-12-05 by fpistm

jpmeijers, 2022-04-18:
Updating to 2.2.0 also fails with a Null Pointer Exception. I was also updating other board definitions today, successfully, so I do not believe the issue is with the Arduino IDE.

```
java.lang.NullPointerException
java.lang.RuntimeException: java.lang.NullPointerException
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$2(ContributionManagerUI.java:175)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.NullPointerException
	at cc.arduino.contributions.packages.ContributionInstaller.remove(ContributionInstaller.java:276)
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$2(ContributionManagerUI.java:170)
	... 1 more
```

[image]

fpistm, 2022-04-19:
Update your arduino ide. 1.8.13 is old and some changes have been done to the json specs.