stm32duino/Arduino_Core_STM32 #1600: Add SPI slave example

brainelectronics, 2021-12-15:
The usage of the SPI library in slave mode is not straight forward, maybe even not implemented.

All examples show how to transmit data FROM a STM32 to some Arduino board (which is not a STM32). In some usecases the STM32 should also be able to be the controlled device to provide data to the SPI master or perform low level task.

brainelectronics, 2021-12-16:
_Also already requested by @AnHardt in https://github.com/stm32duino/Arduino_Core_STM32/issues/257#issuecomment-553774770_

ABOSTM, 2021-12-16:
Hi @brainelectronics ,
In fact the point is not to have an SPI slave example:
the problem is that SPI library from this core, supports only master and doesn't support slave,
and it is the same for Arduino SPI library.

Note that you can still make use of STM32Cube HAL/LL API within Arduino sketch.
Thus you can look at STM32Cube HAL/LL SPI examples.

brainelectronics, 2021-12-16:
@ABOSTM thanks for clarification.
Meanwhile implemented it with the HAL SPI examples for the project I'm currently working on, just thought as some extension of the SPI lib is currently ongoing it might fit in there as well.

Big thank you for this great package and your contributuion, that is maybe not often enough said 👍

closed 2021-12-16 by brainelectronics