stm32duino/Arduino_Core_STM32 #1601: Call of overloaded attachInterrupt is ambiguous with a lambda callback

NepEgor, 2021-12-16:
When calling attachInterrupt with a lambda callback instead of a function I got the following error:
```c++
error: call of overloaded 'attachInterrupt(const uint8_t&, setup()::<lambda()>, int)' is ambiguous
   32 |     attachInterrupt(pin_trackpad_clock[0], [](){trackpad[0].int_on_clock();}, FALLING);
...
C:\Users\<user>\.platformio\packages\framework-arduinoststm32\cores\arduino/WInterrupts.h:28:8: note: candidate: 'void attachInterrupt(uint32_t, callback_function_t, uint32_t)'
   28 |   void attachInterrupt(uint32_t pin, callback_function_t callback, uint32_t mode);
      |        ^~~~~~~~~~~~~~~
C:\Users\<user>\.platformio\packages\framework-arduinoststm32\cores\arduino/WInterrupts.h:32:6: note: candidate: 'void attachInterrupt(uint32_t, void (*)(), uint32_t)'
   32 | void attachInterrupt(uint32_t pin, void (*callback)(void), uint32_t mode);
      |      ^~~~~~~~~~~~~~~
```
Header file WInterrupts.h indeed has 2 prototypes of this function and compiler can't decide which one to use with a lambda.
When the prototype with a function pointer (line 32) is removed, project compiles and works no matter attachInterrupt is called with a lambda or a function pointer.

Please note that Arduino version of attachInterrupt only has one prototype with a function pointer
and esp32 version has one prototype with std::function<void(void)>.

ABOSTM, 2021-12-17:
Hi @NepEgor ,

> Header file WInterrupts.h indeed has 2 prototypes of this function and compiler can't decide which one to use with a lambda.

Yes I agree. As far as I know, lambda in by default not typed, and so both prototypes match your `attachInterrupt()`
So using an  untyped Lambda is not supported. But I don't consider this as a bug.

> When the prototype with a function pointer (line 32) is removed, project compiles and works no matter attachInterrupt is called with a lambda or a function pointer.

Removing the function pointer (line 32), is not a good solution for us because it prevent user to use interrupts API from C files.
And I don't see obvious solution to support untyped lambda for both C and C++.

But you can help compiler to decide which prototype to use,
you can force the type `callback_function_t ` of your lambda:
for example:   `attachInterrupt(digitalPinToInterrupt(USER_BTN), [] () -> callback_function_t {  IT = true;}, RISING);`
Other solution, of course use a standard function instead of a lambda.

NepEgor, 2021-12-17:
Hi, @ABOSTM,

> So using an untyped Lambda is not supported.

It works on Arduino and esp32 frameworks so I expected the same behavior from stm32.

> use interrupts API from C files

I tried to call attachInterrupt from main.c which compiled with gcc as a C file and got the following linker error. I might be doing something wrong as I usually do not use C in PlatformIO. Also I tried to call it from a separate .c file with the same outcome.
```c
c:/users/<user>/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/blackpill_f411ce/src/main.c.o: in function `setup':
main.c:(.text.setup+0x6): undefined reference to `attachInterrupt'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\blackpill_f411ce\firmware.elf] Error 1
```

> you can force the type callback_function_t of your lambda

Tried doing this as well and received the following warning. I think compiler expects this lambda to return callback_function_t and not be callback_function_t.
```c++
src\main.cpp:10:80: warning: no return statement in function returning non-void [-Wreturn-type]
   10 |     attachInterrupt(1, []() -> callback_function_t {Serial.println("Callback");}, FALLING);
      |                                                                                ^
```

```c++
attachInterrupt(1, (callback_function_t)([](){Serial.println("Callback");}), FALLING);
```
This works though. But now this wont works with Arduino or esp32.
```c++
attachInterrupt(1, (void(*)())([](){Serial.println("Callback");}), FALLING);
```
This works with stm32, Arduino and esp32. But it looks awful!

> Other solution, of course use a standard function instead of a lambda.

Well I was using functions before but decided to switch to lambda because they just look neater. :)

> and esp32 version has one prototype with std::function<void(void)>

I was wrong here. esp32 has prototypes for both void(*)() and std::function<void(void)>
Please see [https://github.com/espressif/arduino-esp32/pull/1728]()
and [https://github.com/esp8266/Arduino/pull/2745]()
The second pull request is from esp8266 framework where functional interrupts were implemented originally.
I suppose something similar could be implemented for stm32 as well.

ABOSTM, 2022-01-06:
@NepEgor,
After analysis, it appears that pure C prototypes are an inheritance of former pure C implementation.
And currently this is not working anymore, compilation fails when `attachInterrupt() i`s invoked from *.c file.
So I removed C prototype, and keep only C++ ones in PR #1617.
Thus lambda are supported.

PR #1617 (2022-01-07): Interrupts: remove C prototype and keep only C++ prototypes
**Summary**
Interrupts: remove C prototype and keep only C++ prototypes

attachInterrupt() C prototypes was not usable from C source file
(inheritance of former pure C implementation), and causes ambiguity
when used with lambda function as callback
Fixes #1601
commit ea9a82e4 (2022-01-07): Interrupts: remove C prototype and keep only C++ prototypes

attachInterrupt() C prototypes was not usable from C source file
(inheritance of former pure C implementation), and causes ambiguity
when used with lambda function as callback
Fixes #1601
closed 2022-01-07 by fpistm