stm32duino/Arduino_Core_STM32 #1621: I am following your steps exactly but I cannot get the E5 board to show up!

ramimehyar, 2022-01-16:
On the wiki you mention:

 git checkout seeed_lora_e5_mini ... does this do anything?

fpistm, 2022-01-16:
> On the wiki you mention:
> 
> git checkout seeed_lora_e5_mini ... does this do anything?

Honestly don't know what you mention. Could you point where you saw this, please?

ramimehyar, 2022-01-16:
I want to use this branch to be able to add the lora e5 dev board to the stm32 core package! I followed all instructions using my gitbash, no success!

fpistm, 2022-01-16:
> I want to use this branch to be able to add the lora e5 dev board to the stm32 core package! I followed all instructions using my gitbash, no success!

Which branch? This core has no e5 dev board support so explain us what you try to achieve else we could not help. I suspect you follow a wiki outside of this organisation.

fpistm, 2022-01-16:
Closed as not related to the official wiki. See #1622

closed 2022-01-16 by fpistm