stm32duino/Arduino_Core_STM32 #1622: exec: "/bin/arm-none-eabi-g++": file does not exist

ramimehyar, 2022-01-16:
I did everything as mentioned and i can get the e5 board, but i get while compiling:

exec: "/bin/arm-none-eabi-g++": file does not exist
Error compiling for board Generic STM32WL series.

Please help me get past this.

fpistm, 2022-01-16:
Could you be more precise? Which core version, IDE version? Selected options in the menu? ....
All relevant information would help to solve your issuel.  And what is the e5 board?

ramimehyar, 2022-01-16:
in the wiki of bertrik, he says to create the directory using the option -f which is not recognized by my gitbash.

like this:   mkdir -f stm32

so I do create the stm32 dir without -f option.

ramimehyar, 2022-01-16:
And when i try to execute: 
$ git checkout seeed_lora_e5_mini

I get:
User@DESKTOP-63Q81U9 MINGW64 /c/Program Files (x86)/Arduino/hardware/stm32
$ git checkout seeed_lora_e5_mini
fatal: not a git repository (or any of the parent directories): .git

ramimehyar, 2022-01-16:
Please help me, i am not able to add the fork

fpistm, 2022-01-16:
So as stated in your other issue you try to follow instructions outside of this organisation. Therefore you should ask support to bertrik.

closed 2022-01-16 by fpistm

fpistm, 2022-01-17:
For ref: https://revspace.nl/LoRa-E5-mini

As his fork is not up to date then there is a misalignment with the installed tool (here the arm none eabi gcc). 
You have installed the latest core version 2.2.0 which install the arm-none-eabi-gcc 10.3.1-2.3   while his fork is based on previous version 10.2.1-1.1. That's why you get this error. Simply install  the 2.1.0 instead of the 2.2.0 and it should works.

But as stated if you met another issue please raise an issue on bertrik GH: https://github.com/bertrik/Arduino_Core_STM32/

ramimehyar, 2022-01-17:
Many thanks, you really helped me! I understand, if i had issues ill ask
him there.

Regards,