stm32duino/Arduino_Core_STM32 #1633: Error: failed to download Segment[0] and Error: failed to download the File

Julie97434, 2022-01-24:
Hi, 

I try to make the LED of the LoRaE5-mini blink. 
For that I followed the tutorial: [https://revspace.nl/LoRa-E5-mini](url)

It uses arduino code to flash the LoRaE5 mini card which is connected with an ST-Link V2. 
The code for blinking the LED is as follows:
```
#include <Arduino.h>

void setup(void)
{
    pinMode(PB5, OUTPUT);
    Serial.begin(9600);
}

void loop(void)
{
    digitalWrite(PB5, (millis() / 500) % 2);
}
```
I followed all the steps indicated in this tutorial, for the ST-link connection I use a NUCLEO-L073rz, for which I took care to remove the 2 jumpers so that the ST-link connection is autonomous from the rest of the board.  

But when I try to upload the program this appears on the arduino terminal : 
```
      -------------------------------------------------------------------
                        STM32CubeProgrammer v2.8.0                  
      -------------------------------------------------------------------

ST-LINK SN  : 066FFF353638425043084926
ST-LINK FW  : V2J38M27
Board       : NUCLEO-L073RZ
Voltage     : 3,25V
SWD freq    : 4000 KHz
Connect mode: Under Reset
Reset mode  : Hardware reset
Device ID   : 0x497
Revision ID : Rev Z
Device name : STM32WLxx
Flash size  : 256 KBytes
Device type : MCU
Device CPU  : Cortex-M4
BL Version  : 0xc2

Memory Programming ...
Opening and parsing file: blinky.ino.bin
  File          : blinky.ino.bin
  Size          : 17644 Bytes
  Address       : 0x08000000 

Erasing memory corresponding to segment 0:
Erasing internal memory sectors [0 8]
Download in Progress:

Error: failed to download Segment[0]
Error: failed to download the File

RUNNING Program ... 
  Address:      : 0x8000000
Warning: failed to run application!
Warning: The core is kept under Reset!
Error: Start operation failed
```
I have two download errors (segment and file) and I noticed that the address in the RUNNING Program section does not correspond to the memory address, there is a 0 missing before the 8 ... 

I don't know how to solve this problem, if someone can help me please.

ABOSTM, 2022-01-25:
Hi @Julie97434,
This board is not officially supported by this stm32duino core (it is a fork) and I don't have this board.
Nevertheless, looking at error message, it looks like STM32CubeProgrammer detects properly ST-link (Board       : NUCLEO-L073RZ), also ST-link version seems quite a recent (ST-LINK FW  : V2J38M27), and finally ST-link detects properly the chip to flash (Device name : STM32WLxx).
So it looks like all connection with STlink are OK, but for some reason, the flash cannot be erased.
... maybe any flash protection or security options you play with (Like Option Bytes , ...) ?

I guess error for file is a consequence of segment error.
Missing 0 in front of 8 is not an issue (0x08000000 == 0x8000000)

As it seems you have the same error with  STM32CubeProgrammer, I would suggest to address this issue in ST community:
https://community.st.com/s/topic/0TO0X000000BX4uWAG/stm32cubeprogrammer

closed 2022-01-25 by ABOSTM

Julie97434, 2022-01-25:
@ABOSTM It was indeed a security error, in STM32CubePro > Option bytes > Read Out protection was in `BB` mode instead of `AA` ... now everything works thanks !!!