stm32duino/Arduino_Core_STM32 #1645: Internal pullup/pulldown for GPIO does not work on STM32L051C8Tx

mkals, 2022-02-04:
I am unable to activate pullup or pulldown resistors for GPIO pins on the STM32L051C8Tx core. The pins remain floating. I have made a minimal sketch, where I only try setting this (see below). I know I am measuring the right pins, as checked by using OUTPUT mode on them. Any suggestions as to what the issue could be? 

```
#define LED_PIN PA_13
#define DATA_PIN PB_7 // PB_6 tried as well

void setup() {

  // digitalWrite(DATA_PIN, HIGH); // tried using this first, but no difference
  pinMode(DATA_PIN, INPUT_PULLUP); // no difference for INPUT_PULLDOWN

  //For testing I am measuring the right pin 
  // pinMode(DATA_PIN, OUTPUT); 
  // digitalWrite(DATA_PIN, LOW);
  
  pinMode(LED_PIN, OUTPUT);
}
```

I am using the latest 2.2.0 version of the core.

Thanks!

fpistm, 2022-02-04:
Hi @mkals 
In fact you use the PinName (PY_n) instead of the pin number (PYn).

```patch
- #define LED_PIN PA_13
- #define DATA_PIN PB_7 // PB_6 tried as well
+ #define LED_PIN PA13
+ #define DATA_PIN PB7 // PB6 tried as well
```

mkals, 2022-02-05:
Thank you for catching that, my bad!

closed 2022-02-05 by mkals