stm32duino/Arduino_Core_STM32 #1661: How to avoid I2C bus stuck after a board reset

cparata, 2022-02-16:
I noticed that sometimes if you reset the board during an I2C transaction the I2C bus remains stuck.
A possible solution that I found is to send a bit-banged STOP sequence before the I2C begin. Maybe it would be better to include this procedure in the begin implementation or in a dedicated Wire API to improve the stability of the library.
You can find below the bit-banged STOP sequence that normally I send to unlock the I2C bus before the I2C begin:

```
pinMode(SCL_PIN, OUTPUT);
pinMode(SDA_PIN, OUTPUT);

for (int i = 0; i<10; i++) {
  digitalWrite(SDA_PIN, LOW);
  delay(3);
  digitalWrite(SCL_PIN, HIGH);
  delay(3);
  digitalWrite(SDA_PIN, HIGH);
  delay(3);
  digitalWrite(SCL_PIN, LOW);
  delay(3);
}

pinMode(SDA_PIN, INPUT);
pinMode(SCL_PIN, INPUT);
``` 

Best Regards,
Carlo

ABOSTM, 2022-02-17:
Hi Carlo, thanks for this sharing.
Any reason to send STOP 10 times in loop ?

cparata, 2022-02-17:
Hi @ABOSTM ,
no in particular. I just try to send the "STOP" sequence 10 times to be sure that all the devices on the I2C bus are able to "see" it. Sometimes the devices take some time to be alive on the bus. But it is just an example. Maybe we can reduce the number of loops.
Best Regards,
Carlo

PR #1663 (2022-02-28): Wire: implement STOP sequence to recover I2C bus.
**Summary**
Wire: implement STOP sequence to recover I2C bus.

Useful in case of bus stuck after a reset for example
Fixes #1661
commit 51fc604a (2022-02-28): Wire: implement Clear Bus sequence to recover I2C bus.
Source:
https://www.nxp.com/docs/en/user-guide/UM10204.pdf
https://bits4device.wordpress.com/2017/07/28/i2c-bus-recovery/

Useful in case of bus stuck after a reset for example
Fixes #1661
closed 2022-02-28 by fpistm