stm32duino/Arduino_Core_STM32 #1664: Compilation error (<bits/c++config.h> missing), running in Portable mode

wowtah, 2022-02-23:
**Describe the bug**
I am running a fresh Arduino IDE 1.8.19 in Portable mode. Trying to compile (an empty) sketch for STM32F4 board, compilation fails with this error:

```
In file included from d:\my\long\path\starts\with\this\arduino-1.8.19\portable\packages\stmicroelectronics\tools\xpack-arm-none-eabi-gcc\10.2.1-1.1\arm-none-eabi\include\c++\10.2.1\stdlib.h:36,
                 from d:\my\long\path\starts\with\this\arduino-1.8.19\portable\packages\STMicroelectronics\hardware\stm32\2.1.0\cores\arduino/wiring.h:24,
                 from d:\my\long\path\starts\with\this\arduino-1.8.19\portable\packages\STMicroelectronics\hardware\stm32\2.1.0\cores\arduino/Arduino.h:36,
                 from sketch\sketch_feb23a.ino.cpp:1:
d:\my\long\path\starts\with\this\arduino-1.8.19\portable\packages\stmicroelectronics\tools\xpack-arm-none-eabi-gcc\10.2.1-1.1\arm-none-eabi\include\c++\10.2.1\cstdlib:41:10: fatal error: bits/c++config.h: No such file or directory
   41 | #include <bits/c++config.h>
      |          ^~~~~~~~~~~~~~~~~~
compilation terminated.
exit status 1
Error compiling for board Generic STM32F4 series.
```

**To Reproduce**
1. Download Arduino IDE (1.8.19) as ZIP, for Windows
2. Extract zip, create folder 'portable' (in extracted folder) to make it into a portable Arduino IDE
3. Start Arduino.exe
4. Preferences -> add Boards manager URL: https://github.com/stm32duino/BoardManagerFiles/raw/main/package_stmicroelectronics_index.json
5. Go to Boards manager, Install 'STM32 MCU based boards' version 2.2.0
6. Switch board to 'Generic STM32F4 series' -> 'STM32F411CEUx'
7. Compile the (empty) sketch
8. Error (see above) occurs

**Expected behavior**
I expect a clean install to compile an empty sketch :) 

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Arduino IDE version: 1.8.19
 - STM32 core version: 2.2.0 (problem also occured with 2.1.0)
 - Tools menu settings if not the default: 'Generic STM32F4 series' -> 'STM32F411CEUx'

**Board (please complete the following information):**
 - Name: STM32F411CEU6

fpistm, 2022-02-23:
Hi @wowtah 
This errors comes from an issue with the long path name.
Try to shorten the path where Arduino IDE is installed and it should works.
I have no other solution for you as it is linked to the IDE on Windows.

Related topic:
https://www.stm32duino.com/viewtopic.php?t=1160
https://github.com/arduino/Arduino/issues/10754 (anyway this not solve all the issue).... 😕

wowtah, 2022-02-23:
Using a shorter path fixed it :) thanks!

closed 2022-02-23 by fpistm