stm32duino/Arduino_Core_STM32 #1668: Arduino IDE 2.0 rc4, upload is done but nothing happen

gbr1, 2022-03-03:
**Describe the bug**
On MacOS 12.2.1 and Arduino IDE 2.0 rc4, I tested the blink sketch on a Nucleo L152 and a Nucleo F401. Everything seems to be ok and upload starts as mass storage.  On board stlink blinks red/green on stlink and stops on green. Firmware doesn't run. Tested also to plug/unplug and reset the nucleo.

**To Reproduce**
Open blink example on Arduino IDE 2.0 rc4 and try to upload on Nucleo F401/ L152

Steps to reproduce the behavior:
1. Upload blink example
2. Onboard stlink blinks red/green and stops on green
3. Nothing happen (firmware doesn't start)

**Expected behavior**
The board should start blinking.

**Desktop (please complete the following information):**
 - OS: MacOS 12.2.1
 - Arduino IDE version: 2.0 rc4
 - STM32 core version: 2.2.0
 - Tools menu settings if not the default: Newlib Standard
 - Upload method: Mass Storage

**Board (please complete the following information):**
 - Name: Nucleo F401RE
 - Hardware Revision: Rev C
and:
 - Name: Nucleo L152RE
 - Hardware Revision: Rev C

fpistm, 2022-03-03:
Hi,

You can try to update the STLink firmware (using STM32CubeProgrammer). 
But it seems more an issue with Arduino IDE 2.0 RC4.
It works properly with Arduino IDE 1.8.19. Even with arduino-cli 0.21.1.
You should submit an issue to the IDE especially since it is a RCx.

closed 2022-03-03 by fpistm

gbr1, 2022-03-03:
@fpistm thank you for suggestion! Updating stlink firmware gets it working!

So now it is possible to work on stm32 using ide 2.0

fpistm, 2022-03-03:
Nice! Good news!  Thanks for the feedback.