stm32duino/Arduino_Core_STM32 #1669: Exposing get_adc_channel function

askuric, 2022-03-04:
**Is your feature request/improvement related to a problem? Please describe.**
It would be beneficial for developers to be able to get the ADC channel of their pin. 
You have already developed the code for that in your `analog.cpp` but the functions are private (not in the `analog.h`)

**Describe the solution you'd like**

If it is possible to add the line 
```cpp
uint32_t get_adc_channel(PinName pin, uint32_t *bank)
```
to the  `analog.h`

**Describe alternatives you've considered**

Maybe an api implementation could be extended to support such requests, but as you have already implemented this functionality it is a shame not to provide it to the users :D

**Additional context**

My context is SimpleFOClibrary implementation where we need to do injected channel pwm triggered ADC implementation in a generic manner and we will need to determine the `ADC_CHANNEL_x` channel  and `ADCx` instance from the pin name.

If you'd like I can do a quick PR for this.

ABOSTM, 2022-03-08:
Hi @askuric,
This is a community project, and your proposal sounds good,
So yes you are welcome to propose a PR :+1:

fpistm, 2022-03-16:
HI 
Any update on this ?

fpistm, 2022-04-26:
@askuric 
will you able to do a PR for this?
We prepare a release so if you want to have it in the time frame of the next one it should be fine to get it. Thanks.

PR #1735 (2022-06-08): Review get adc/dac/timer channel functions
This PR exports some getter api:
```
uint32_t get_adc_channel(PinName pin, uint32_t *bank);
uint32_t get_adc_internal_channel(PinName pin);
uint32_t get_dac_channel(PinName pin);
```

Fixes #1669