stm32duino/Arduino_Core_STM32 #1672: `while (!Serial);` hangs on Swan R5 with `CDC (generic 'Serial' supercede U(S)ART)`

zfields, 2022-03-08:
**Describe the bug**
Firmware hangs at `while (!Serial);`.

Here is the implementation:
```cpp
USBSerial::operator bool()
{
  delay(10);
  return dtrState;
}
```

I took a cursory look, and I don't see anywhere `dtrState` is getting set.

Here is the Arduino Board Configuration
[image]

The bug only appears with CDC (generic 'Serial' supercede U(S)ART).

The bug does not reproduce with:
- CDC (no generic 'Serial')
- HID (keyboard and mouse)

**To Reproduce**
Complete source code which can be used to reproduce the issue. Please try to be as generic as possible (no extra code, extra hardware,...)

Steps to reproduce the behavior:
1. Flash this program to the device
   ```c++
   void setup()
   {
       Serial.begin(115200);
       while (!Serial);
       pinMode(LED_BUILTIN, OUTPUT);
   }
   
   void loop()
   {
       Serial.println("Looping!");
       static uint32_t state = LOW;
       state = !state;
       digitalWrite(LED_BUILTIN, state);
       delay(750);
   }
   ```
2. The device will not blink or print "Looping!".

3. Comment out
   ```c++
      while (!Serial);
   ```
4. Reflash device
5. The device will work as expected.

**Expected behavior**
I would expect Serial == true (a.k.a. to be "ready"), if it can send bytes over USB.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Linux
 - Arduino IDE version: 1.8.19
 - STM32 core version: _latest_

**Board (please complete the following information):**
 - Name: Swan R5
 - Hardware Revision:

**Additional context**

VSCode Version: 1.65.1
Commit: 8908a9ca0f221f36507231afb39d2d8d1e182702
Date: 2022-03-08T02:09:03.269Z
Electron: 13.5.2
Chromium: 91.0.4472.164
Node.js: 14.16.0
V8: 9.1.269.39-electron.0
OS: Linux x64 5.16.11-76051611-generic

fpistm, 2022-03-09:
Hi @zfields 
Could you try to disable dtr before the while:
`Serial.dtr(false);`

See #1382

zfields, 2022-03-09:
```c++
void USBSerial::dtr(bool enable)
{
  CDC_enableDTR(enable);
}
```

```c++
void CDC_enableDTR(bool enable)
{
  CDC_DTR_enabled = enable;
  dtrState = true;
}
```

Would it be possible to use a `#define` to set `dtrState = true` if `CDC (generic 'Serial' supercede U(S)ART` is selected as a board option?

fpistm, 2022-03-09:
The `Serial.dtr(bool)` does not set the dtr value only enable or not the dtr feature as not all the terminal use it.

About:  

> I'm curious how it is working with `CDC (no generic 'Serial')` and `HID (keyboard and mouse)`, without having to manually set the DTR state.
and:
> The bug does not reproduce with:
> 
>     * CDC (no generic 'Serial')
> 
>     * HID (keyboard and mouse)

This is normal it does not occurred as Serial is mapped on a Hardware Serial instance in this case (Serialx) not the SerialUSB.

``

So:
> Would it be possible to use a `#define` to set `dtrState = true` if `CDC (generic 'Serial' supercede U(S)ART` is selected as a board option?

No this is not the goal, dtr state is a volatile that can change if the dtr feature is available (default case). If you disable it then it will be always true.

zfields, 2022-03-09:
I don't understand what you mean by the following statement...

> The Serial.dtr(bool) does not set the dtr value only enable or not the dtr feature as not all the terminal use it.

How is this the case?

```c++
void USBSerial::dtr(bool enable)
{
  CDC_enableDTR(enable);
}
```

```c++
void CDC_enableDTR(bool enable)
{
  CDC_DTR_enabled = enable;
  dtrState = true;
}
```

Based on the code above, it appears `Serial.dtr(bool)` _ALWAYS_ sets the DTR value (`dtrState`) to `true`.

fpistm, 2022-03-09:
I've tested your sketch with my Swan board and have no issue.

fpistm, 2022-03-09:
> Based on the code above, it appears `Serial.dtr(bool)` _ALWAYS_ sets the DTR value (`dtrState`) to `true`.

Yes it set it to true when you force to be enable but after that it is updated:
https://github.com/fpistm/Arduino_Core_STM32/blob/67a5298b92140de11a38a2c10869d505eb1c1466/cores/arduino/stm32/usb/cdc/usbd_cdc_if.c#L189

zfields, 2022-03-09:
I'm using the terminal supplied the the Arduino extension in VSCode.

zfields, 2022-03-09:
I feel like we are getting caught up in the details.

Zooming out...

- I can write supply any value to `Serial.dtr()` just to make the API work correctly.
- Bytes can be sent from the device and received by the terminal, even though `!Serial` doesn't evaluate to `false`.
- It sounds like you are suggesting that I write non-portable code `Serial.dtr()` for ?all? STM32 boards, or add this to my code:
  ```cpp
  #if defined(ARDUINO_ARCH_STM32) && (Serial == USBSerial)
  Serial.dtr(false);
  #endif
  ```
- `!Serial` works properly ESP32 or AVR boards.

zfields, 2022-03-09:
Thank you for the detailed response! That really provided some insight. :+1: 

Let me better explain this statement I made earlier:

> Regardless of the call to Serial.dtr(), bytes can be sent from the device and received by the terminal, even though !Serial doesn't evaluate to false.

When you said, "I don't understand your point: as long as the while(!Serial) is not evaluated true and no Serial.print()... are called the device send nothing."

What I meant to say is, I am already able to send bytes (i.e. `Serial.println("Hello, world!");`) and see it in my console. However, when I add `while (!Serial);` (without `Serial.dtr(true|false)`) `!Serial` will _NEVER_ evaluate to `false` and let me to move on to the next line of code.

> That's why I asked you if your terminal managed or not the linestate and you didn't answer it.

I'm sorry about that, I don't know how to check. I was hoping that by sharing the fact it does work without checking `!Serial` would be enough to answer your question. If you can tell me how to check, I will definitely look for you.

Thanks for taking the time to clarify. Tell me how to collect the information you need and I will gladly do it.

fpistm, 2022-03-10:
Based on your input it seems the terminal does not manage the DTR which explain why it does not work.
Looking at the VSCode Arduino extension GH issue and seems loot of them are linked to DTR issue management:
https://github.com/microsoft/vscode-arduino/issues/1458
https://github.com/microsoft/vscode-arduino/issues/1285
https://github.com/microsoft/vscode-arduino/issues/867

Anyway, I made a commit to fix issue about the dtrState force to true by default when using the `dtr(bool)` api (https://github.com/stm32duino/Arduino_Core_STM32/commit/6187a43c8be6cf6353fb51afbfc429ce5d2884e5).
Could you try on your side to use a terminal with DTR support to confirm it works as expected with your default sketch?
For example using the Serial monitor of the Arduino IDE.

fpistm, 2022-03-11:
Hi @zfields 
Any update on this ?

zfields, 2022-03-11:
Not yet, I got hung up on the other issue. My expectation is that it would fail, because the line that you removed updates the value that is specifically returned by `USBSerial();`.

`USBSerial` must return `true` in order to fail out of `while (!Serial);`

fpistm, 2022-03-11:
If it fails it is because the serial console does not managed the DTR (Data Terminal Ready).
From core side this is what is it expected. Moreover the goal of the `while(!Serial);` is to ensure the terminal is ready to receive.

With Arduino IDE Serial monitor it should work.

zfields, 2022-03-11:
I will test with the Arduino IDE and let you know.

zfields, 2022-03-14:
I haven't been able to confirm this, because of issue https://github.com/stm32duino/Arduino_Tools/issues/84.

fpistm, 2022-03-14:
Hi @zfields 
all that you said confirm this.
I've made several test and it works as expected (only the `Serial.dtr(true)` which forced by default to true which is now fixed). 
If you use the while(!Serial) and your terminal does not manage the DTR then it's up to you to disable the DTR management if you want keep this terminal or change the terminal.

zfields, 2022-03-14:
I've figured it out, and it works perfectly. The linked issue :point_up_2: threw me off.

Once I flashed using DFU instead of SWD, then the Virtual COM Port appeared correctly. This is a necessary prerequisite to this function working as expected. In Arduino, the program will hang until I open the **Serial Monitor** (`Ctrl+Shift+M`), and in VSCode, the program will hang until I open the connection using the Serial Monitor (in the Terminal window area).

This behaves _EXACTLY_ as I had hoped! It appears I was getting tripped up by the board being misconfigured.