stm32duino/Arduino_Core_STM32 #1682: Tools Dependencie xpack-arm-none-eabi-gcc version 10.3.1-2.3 does not exist

jfm-made, 2022-03-24:
**Describe the bug**
toolsDependencie xpack-arm-none-eabi-gcc version 10.3.1-2.3 does not exist.

**To Reproduce**
Add "https://github.com/stm32duino/BoardManagerFiles/raw/main/package_stmicroelectronics_index.json" to Arduino IDE board manager urls.
Install version 2.2.0 with the Arduino IDE. It will fail downloading xpack-arm-none-eabi-gcc version 10.3.1-2.3 with status 404.

**Expected behavior**
An existing version of xpack-arm-none-eabi-gcc is downloaded.

**Screenshots**
[image]

**Desktop:**
 - OS: Windows x64 (Tested also on Intel Mac and Apple Silicon, with latest version of Arduino IDE)
 - Arduino IDE version: 1.8.10

fpistm, 2022-03-24:
Hi,
It should be fixed. See https://github.com/xpack-dev-tools/arm-none-eabi-gcc-xpack/issues/17.

Hope sha256 is the same.

JoanMCD, 2022-03-25:
The URL is still not working.

fpistm, 2022-03-25:
> 

I've tested and it works:
https://github.com/xpack-dev-tools/arm-none-eabi-gcc-xpack/releases/download/v10.3.1-2.3/xpack-arm-none-eabi-gcc-10.3.1-2.3-win32-ia32.zip

Maybe Arduino cache issue or proxy

JoanMCD, 2022-03-25:
I reflect to this URL: "https://github.com/stm32duino/BoardManagerFiles/raw/main/package_stmicroelectronics_index.json"
Ending with .json and needed in the Arduino IDE

fpistm, 2022-03-25:
I don't understand your previous comment. The json  file is the entry point then depending of the version you installed you get list of files to download. The xpack one was removed by mistake by the owner and it is now available.

JoanMCD, 2022-03-25:
Yes - but the json isn't available -> click it.

fpistm, 2022-03-25:
> Yes - but the json isn't available -> click it.

It is available. Moreover your point concern a package reference in this json. 
When I click on your link it appears properly.

JoanMCD, 2022-03-25:
Oh? Yes you're right. Checked it by VPN and it's working. Hipee :)
 Sorry for interrupting and thank you for your time. It must be a local problem on my side. 
regards Joan

closed 2022-03-25 by fpistm