stm32duino/Arduino_Core_STM32 #1692: [新变体]looking forward the support for Nucleo h72zg!

alin2585, 2022-04-16:
nucleo h723zg is a excellent board. looking forward your support for it.

closed 2022-04-16 by alin2585

uzi18, 2022-04-16:
@alin2585 link to this excellent board?

alin2585, 2022-04-16:
https://www.st.com/en/microcontrollers-microprocessors/stm32h723zg.html
sorry , do you mean this kind of link?

alin2585, 2022-04-16:
https://www.st.com/en/evaluation-tools/nucleo-h723zg.html

alin2585, 2022-04-17:
@fpistm I attempted to upload my help.I don't know if it is right.please tell me the problems in it and I will edit it right now.
https://github.com/alin2585/stm32dunio-support-for-h723zg

fpistm, 2022-04-19:
Hi @alin2585 
you should have a look on this:
https://github.com/stm32duino/wiki/wiki/Add-a-new-variant-%28board%29

If you add a specific board (not a generic), you can see ow it is added for the Nucleo H743ZI:
https://github.com/stm32duino/Arduino_Core_STM32/tree/main/variants/STM32H7xx/H742Z(G-I)T_H743Z(G-I)T_H747A(G-I)I_H747I(G-I)T_H750ZBT_H753ZIT_H757AII_H757IIT

I guess you talk about this one: https://www.st.com/en/evaluation-tools/nucleo-h723zg.html