stm32duino/Arduino_Core_STM32 #1693: Getting "ERROR: invalid transmit pin number" on correct UART1 pins of Nucleo-F446RE

N0rbert, 2022-04-17:
**Describe the bug**
Getting "ERROR: invalid transmit pin number" on correct UART1 pins of Nucleo-F446RE.

**To Reproduce**
I'm using Arduino 1.8.19 with Arduino Core for STM32 2.2.0.

Below is the minimal reproducible example code which I run on Nucleo-F446RE:

```
#include <SoftwareSerial.h>
#define SSerialRX        2
#define SSerialTX        8
#define SSerialTxControl 3
 
#define RS485Transmit    HIGH
#define RS485Receive     LOW

const char my_cmd[] = {'H', 'I'};

SoftwareSerial RS485Serial(SSerialRX, SSerialTX);
 
void setup() {
  Serial.begin(9600);
  Serial.println("RS485 demo");
  pinMode(SSerialTxControl, OUTPUT);
 
  digitalWrite(SSerialTxControl, RS485Receive);
  RS485Serial.begin(9600);   // set the data rate .

  delay(1000);

  digitalWrite(SSerialTxControl, RS485Transmit);

  Serial.println("Sending our text...");
  Serial.println(my_cmd);

  for (int i=0; i<2; i++)
    RS485Serial.write(&my_cmd[i]);

  digitalWrite(SSerialTxControl, RS485Receive);
}

void loop() {
}
```

with MAX485 connected to it. While running it prints the following error output to the Serial Monitor:

>```
>RS485 demo
>Sending our text...
>HIERROR: invalid transmit pin number
>```

I find this behavior unexpected and erroneous. 

Steps to reproduce the behavior:
1. Compile
2. Upload to board
3. See error

**Expected behavior**
No error.

**Desktop (please complete the following information):**
 - OS: Ubuntu MATE 18.04.5. LTS
 - Arduino IDE version: 1.8.19
 - STM32 core version: 2.2.0
 - Upload method: Mass Storage

**Board (please complete the following information):**
 - Name: Nucleo-F446RE
 - Hardware Revision: MB1136 Rev. C
 - Extra hardware used if any: MAX485 on UART1 pins

**Additional context**
The same hardware configuration works normally on Mbed OS 5 with the same pins assigned to UART1.

fpistm, 2022-04-19:
Hi,
as a first guess, `println` should require a nul terminated string.

```
const char my_cmd[] = {'H', 'I'; '\0'}; // or "HI";
....
  Serial.println(my_cmd);
```

Could you try to fix this first?

N0rbert, 2022-04-19:
Holy molly, it really helped. Thanks!

closed 2022-04-19 by N0rbert