stm32duino/Arduino_Core_STM32 #1698: STM32F401CC BOARD FREEZES WHEN rtc.setClockSource(STM32RTC::LSE_CLOCK); IS CALLED

augustinenwafor1997, 2022-04-26:
My code involves using the stm rtc library and works fine on the F103 (bluepill) board. But when I upload the coad to the F401 board, it freezes in the setup. I have attached the code.`#include <STM32RTC.h>
#include <TimeLib.h>
STM32RTC& rtc = STM32RTC::getInstance();
void setup(void)
{
  char valve_A[4] = {valve_1A, valve_2A, valve_3A, valve_4A};
  char valve_B[4] = {valve_1B, valve_2B, valve_3B, valve_4B};
  for (int i = 0; i < sizeof(valve_A); i++)
  {
    pinMode(valve_A[i], OUTPUT);
    pinMode(valve_B[i], OUTPUT);
  }
  menubutton_Init(PB13, PB14, PB12, PA8, PB15);
  pinMode(lcdBacklight, OUTPUT);
  pinMode(FS1, INPUT_PULLUP);
  pinMode(FS2, INPUT_PULLUP);
  pinMode(pumpPin, OUTPUT);
  digitalWrite(PB3, HIGH);
  digitalWrite(lcdBacklight, LOW);
  pinMode(PC13, OUTPUT);
  digitalWrite(PC13, HIGH);
  rtc.setClockSource(STM32RTC::LSE_CLOCK);
  rtc.begin();
}`

ABOSTM, 2022-04-26:
Hi @augustinenwafor1997
Which F401 board are you talking about ? Nucleo ? Which revision of the board ?
Are you sure there is an LSE on this board?
Presence of LSE on Nucleo F401 depends on board revision.

I tested example 'simpleRTCAlarm' on my Nucleo_F401RE (Rev MB1136 C-04  with LSE), I uncommented the line to use LSE and it works fine.
Can you test this sketch and check whether it works: 
you can test both LSE and LSI.
 
Can you, for test purpose, try to replace LSE by LSI in your own sketch and check if it is working:
`rtc.setClockSource(STM32RTC::LSI_CLOCK);`

augustinenwafor1997, 2022-04-26:
I am using the generic blackpill f401ccu6. It has an LSE on board.
I will replace LSE with LSI and get back to you

augustinenwafor1997, 2022-04-27:
> I am using the generic blackpill f401ccu6. It has an LSE on board. I will replace LSE with LSI and get back to you

Hi @ABOSTM 
I tried the suggested line and something interesting happened. When I use the line:
`rtc.setClockSource(STM32RTC::LSI_CLOCK);` the code does not freeze in setup and proceeds to run correctly but when I comment it which should still select LSI by default as written in the documentation, the code freezes in setup same as when I select the LSE as clock source. 
Do you have any other suggestion as to what I should do?
I think I will try replacing the 32khz crystal with the one from my old blue pill.

fpistm, 2022-04-28:
Hi @augustinenwafor1997 
Could you tell us more about your setup?
Arduino IDE version, core version, STM32RTC library version, Host OS,...? All relevant information can help to understand your issue.

ABOSTM, 2022-04-28:
Hi @augustinenwafor1997,
Yes changing 32khz crystal is a good idea.

I agree that commenting setClockSource() should still select LSI by default.
So please provide all information requested by @fpistm

Also can you test the same with example 'simpleRTCAlarm'
because I know it is working on my side, and it avoids to be polluted by anything else than RTC.

augustinenwafor1997, 2022-04-28:
Thank you all for all the feedback so far.
@fpistm Arduino version: 1.8.12
Core version: 1.9.0
STM32RTC library version: 1.0.3
Programmer: ST Link V2
Board: STM32F401CCU6 BLACKPILL (GENERIC)
IDE: ARDUINO

augustinenwafor1997, 2022-04-28:
@ABOSTM I tried with the 'simpleRTCAlarm' example and it's all still the same. The timer works when LSI is selected as the clock source and freezes when it is commented or when LSE is selected as the clock source. I have attached the code below.
`#include <STM32RTC.h>
#include <U8g2lib.h>
#ifdef U8X8_HAVE_HW_SPI
#include <SPI.h>
#endif
#ifdef U8X8_HAVE_HW_I2C
#include <Wire.h>
#endif
U8G2_ST7567_OS12864_1_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/PA5, /* data=*/PA7, /* cs=*/PA4, /* dc=*/PA3, /* reset=*/PA1);

/* Get the rtc object */
STM32RTC& rtc = STM32RTC::getInstance();

/* Change these values to set the current initial time */
const byte seconds = 0;
const byte minutes = 0;
const byte hours = 16;

/* Change these values to set the current initial date */
const byte day = 25;
const byte month = 9;
const byte year = 15;

void setup()
{
  Serial.begin(9600);

  // Select RTC clock source: LSI_CLOCK, LSE_CLOCK or HSE_CLOCK.
  // By default the LSI is selected as source.
  rtc.setClockSource(STM32RTC::LSI_CLOCK);

  rtc.begin(); // initialize RTC 24H format

  rtc.setTime(hours, minutes, seconds);
  rtc.setDate(day, month, year);

  u8g2.begin();
  u8g2.enableUTF8Print();
  u8g2.firstPage();
  do {
    u8g2.setFont(u8g2_font_fub17_tr);// u8g2_font_fub17_tr
    u8g2.drawStr(10, 40, "COSMOS");
    u8g2.setFont(u8g2_font_helvB08_te);// u8g2_font_fub17_tr
    u8g2.drawStr(54, 50, "automation");
  } while ( u8g2.nextPage() );
  delay(5000);

  rtc.attachInterrupt(alarmMatch);
  rtc.setAlarmDay(day);
  rtc.setAlarmTime(16, 0, 10);
  rtc.enableAlarm(rtc.MATCH_DHHMMSS);
}

void loop()
{

}

void alarmMatch(void *data)
{
  UNUSED(data);
  Serial.println("Alarm Match!");
  u8g2.firstPage();
  do {
    u8g2.setFont(u8g2_font_helvB08_te);// u8g2_font_fub17_tr
    u8g2.drawStr(20, 40, "ALARM MATCH");
  } while ( u8g2.nextPage() );
}`

a picture of my setup is attached
[image]

ABOSTM, 2022-04-28:
> Arduino version: 1.8.12 
> Core version: 1.9.0 
> STM32RTC library version: 1.0.3
> Programmer: ST Link V2 
> Board: STM32F401CCU6 BLACKPILL (GENERIC)
> IDE: ARDUINO

That sound like an relatively old installation, ... many things have changed since.
Can you please update:
* Arduino version: 1.8.19
* Core version: 2.2.0   warning , you need to change Boards Managers URLs for stm32 boards: 
see details https://github.com/stm32duino/wiki/wiki/Getting-Started#installing-stm32-cores
*  STM32RTC library version: 1.2.0

and try again

augustinenwafor1997, 2022-04-28:
I did all the updates but now I am getting a compile error saying "fatal error: core_cm4.h: No such file or directory
  140 | #include "core_cm4.h""

ABOSTM, 2022-04-28:
It looks like an issue during installation. Like #1317
Can you try to reinstall

JiriBilek, 2022-04-28:
Hi, reading this I reminded of the problem with the "Blue Pill" and intermittent failures of RTC when there was a connection from the OSC32 pins to the breakout board. Apparently, this changed the capacity between the crystal and ground and the oscillator then failed. As far as I remember the solution was to remove pin headers for the two pins connected to OSC32.
It's just an idea, I haven't tried it myself.

augustinenwafor1997, 2022-04-28:
> Hi, reading this it reminded me the problem with the "Blue Pill" and intermittent failures of RTC when there was a connection from the OSC32 pins to the breakout board. Apparently, this changed the capacitance between the crystal and the ground and the oscillator then failed. As far as I remember the solution was to remove pin headers for the two pins connected to OSC32. It's just an idea, I haven't tried it myself.

This is a solution to a different problem. I experienced it with the blue pill when I wanted to use it on a breadboard some years back and removing the pins fixed that.

fpistm, 2022-05-10:
I close this issue as it is related to hardware failure. All our tests passed.

closed 2022-05-10 by fpistm